webhdx/PicoBoot #1: Not an issue, but a question

Androxilogin, 2022-06-30:
With the Pi Pico W releasing today, are OTA updates in the works? And maybe, just maybe, online play? It would be cool to see this utilize the Bluetooth capabilities for controllers such as 8bitdo as well once they get the drivers figured out. I'm not sure if these chips provide enough memory for all of these applications but that would be pretty awesome.

Ch0z, 2022-06-30:
> With the Pi Pico W releasing today, are OTA updates in the works? And maybe, just maybe, online play? It would be cool to see this utilize the Bluetooth capabilities for controllers such as 8bitdo [...]

For bluetooth capabilities there is already a project for that [darthcloud/BlueRetro](https://github.com/darthcloud/BlueRetro), I don't think bluetooth control is the goal of this project.

AmyGrrl, 2022-06-30:
I can't wait to see what happens with the Pi Pico W. Like having it auto update itself and Swiss. I would also love to see if we could have custom splash screens/sounds on boot instead of the IPL Boot Menu. You can set how long you want the splash screen to display for before it boots Swiss. Would be nice if we could support animations somehow. But space on the Pico is pretty limited. Anyways. Thanks for this amazing project!

webhdx, 2022-07-01:
Pico W was announced yesterday, I didn't had it in advance to work on WiFi features. I ordered a few Pico W's and will try to provide at least OTA update functionality.

AmyGrrl, 2022-07-01:
That would be pretty amazing to have OTA updates to the PicoBoot and Swiss. I ordered both a Pico & Pico W. I would like to see if you could add support to have a splash screen display and/or play a sound clip on boot instead of seeing the iplboot screen. Have a way to set how long the splash screen displays for. Maybe even support for animated pngs or something like just displaying a bunch of jpgs in a sequence. I know space is limited on the Pico. So it could be a fun challenge to get it all to fit. Thanks for this amazing project!

willsteinmetz, 2022-07-01:
> With the Pi Pico W releasing today, are OTA updates in the works? And maybe, just maybe, online play? It would be cool to see this utilize the Bluetooth capabilities for controllers such as 8bitdo [...]

I wish my software engineering realm overlapped enough for me to understand how to contribute. Bluetooth for the controllers would be amazing (both of my dogs have managed to trip over 12 feet of cable essentially glued to the floor before)! Sadly, all of my background is in web software engineering so not a ton of overlap :(

closed 2022-07-02 by webhdx

hpoom, 2022-07-04:
@Ch0z Regarding [darthcloud/BlueRetro](https://github.com/darthcloud/BlueRetro), it would be nice to only have a single board inside the Gamecube.

Maybe it is about getting BlueRetro to run on a Pi Pico W instead of PicoBoot supporting bluetooth controllers. If that happened would you be able to compile the two together to get a single package to flash to the board?

(I am already running an ESP32 with BlueRetro, and plan to get my PicoBoot installed this weekend. Thanks @webhdx for all you work!)