webhdx/PicoBoot #6: Raspberry Pico Version "W"

NicolasMKZ, 2022-07-03:
I just brought an idea because she has a Wifi card and this idea would make it possible to no longer worry about whether the system is up to date or not.

The wifi card connects the system to the internet and it could allow it to install the latest system on the Pico directly if an update was to be expected.

Its will allow that we don't need to disassemble the console each time a new version comes out and it would make it easier for inexperienced people (like me).

Thank you for thinking of this alternative and looking into it!

NicolasMKZ, 2022-07-03:
And another idea for personalization and file playback of on "swiss" it is possible to use a "microSD breakout card" which would free up maximum space on other devices and contact the whole thing for a  precise reading and writing as needed and it will also allow you to be able to update the system without the risk of flocking the system in memory in the event of a bug (if you program well to read part of the system / code on the device to install).

This does not  I have only one follow-up idea which can be very interesting in my opinion and which are offered for information purposes to help the evolution of the project!

webhdx, 2022-07-03:
Thanks for the ideas. I'll be evaluating possibilities of using Pico W features in the future.

closed 2022-07-03 by webhdx

NicolasMKZ, 2022-07-04:
I advise you to add more labels to the posts "Idea", "Bug", "Other" it will help you not to look for 10 years for a post or an idea 😇👍🏻