webhdx/PicoBoot #7: Panasonic Q support

darkshu007, 2022-07-04:
It would be fantastic if the Picoboot could support the Panasonic Q, a Gamecube from Panasonic with a DVD player. These consoles are incompatible with the Xeno GC, and no other modchip today supports it. The old modchip like Viper and Qoob supported it. I was the first to install a Viper in a Panasonic Q. I know because they updated the firmware at my request to make it work, and I tested it for them and confirmed it worked. I have a few other Q. I would like to mod them using the PicoBoot, which I think is a great project. The Panasonic Q with PicoBoot would be the ultimate GameCube! I would be happy to beta test it for you and contribute by creating a tutorial.

Thank you

NicolasMKZ, 2022-07-04:
Il prend déjà en charge le Panasonic Q j'ai u cette conversation il y a peu sur un forum avec le développeur et c'est enfin la 1er vraiment alternative pour le Panasonic Q je suis vraiment heureux de savoir que c'est enfin possible !

(Et c'est à un prix ultra raisonnable !)

Si tu a besoin de plus d'informations je me suis penché sur le code et sur le fonctionnement du Panasonic Q !

webhdx, 2022-07-04:
Panasonic Q is already supported, install it using DOL-001 diagram.

closed 2022-07-04 by webhdx

darkshu007, 2022-07-06:
> It already supports the Panasonic Q. I had this conversation recently on a forum with the developer, and it's finally the 1st really alternative for the Panasonic Q I'm really happy to know that it's finally possible!
> 
> PS: I don't think there is need for beta test and a schematic because the GameCube Panasonic Q motherboard is the same as GameCube Classic.
> 
> (And it's at an ultra reasonable price!)
> 
> If you need more information, I looked at the code and the operation of the Panasonic Q!

Thank you for the information. I would like it if you could share more information about the Q. I am looking at finding information on the dvd\mp3 mod that I installed on my first Q. It is essentially a Panasonic DVD mod using the Panadvd 2.0, a pic chip, and additional memory chip and caps for pal playback. Let me know if how I can reach you ;)

NicolasMKZ, 2022-07-10:
You can contact me on my Discord:

HinataNekoNya#2811