webhdx/PicoBoot #8: Ability to boot stock bios

Adran-Marit, 2022-07-04:
My understanding is that once the picoboot modchip is installed it will boot ipl.dol (most notably swiss), however I was wondering if there was a way to boot the stock bios with the chip installed without removing the SD2SP2?

Whether this is by not having the ipl.dol and just having b.dol for example to only boot swiss while holding B, and without the button press just boots the normal bios.

Many thanks and cool project, I hope it grows <3

webhdx, 2022-07-04:
Yes. Don't put `ipl.dol` file at all and name Swiss as  one of `x/y/b.dol` files. Then you can load Swiss on demand.

> I just bought a SD gecko, for another 20 bucks out of my pocket, I hope it works with that...

It shouldn't cost you more than $5... Not sure where did you order it 🤔

nex86, 2022-07-04:
> Owner

ordered it on Amazon with 2 day shipping from my country.
Anything cheaper would take weeks to months to ship..

closed 2022-07-05 by webhdx