webhdx/PicoBoot #11: No feed

Yuuki2012, 2022-07-05:
Q: Is everything soldered correctly?
A: Yes. Reflowed everything and even uninstalled, tested (worked without mod), installed again, tested again, on both Pico and GC, but still nothing.

Q: What GC do you have?
A: DOL-001

When turning on the GC, all I get is a black screen. Pico LED does turn on, but nothing happens. (I'm using SD2SP2, FAT32 formatted 4GB card (tested with exFAT as well))

Anyone any ideas?

chase-cobb, 2022-07-05:
I'm having the same issue as well, and it is intermittent. Sometimes it will boot as intended, and sometimes I get a black screen. Same is true with or without SD2SP2.

All connections double-checked. Will change to heavier wiring,shorten and respond here if ticket is still open.

kjuno, 2022-07-05:
its very important to have stable solder connections, had to reflow everything to get it working

chase-cobb, 2022-07-05:
> 

My issue is not one of poor solder joints. I suspect it is wire length/gauge.

Yuuki2012, 2022-07-05:
It might be the same for me. Will see when I have time in a few days.

If anyone else can check in the mean time, much appreciated.

chase-cobb, 2022-07-05:
After swapping to shorter wires that are slightly heavier, it is functioning reliably. Went from about 6 inches of wire to about 4 inches.

EtienneCK, 2022-07-05:
I got the very same issue yesterday, but I definitely had too thin and long wires, will redo the work tonight...

maniaco007, 2022-07-06:
I will redo the connections using smaller and thicker cables. As soon as you do, I'll let you know the result.

maniaco007, 2022-07-06:
I redid the connections with thicker and smaller cables and I was successful!

There are alternative solder points for GP4 and GP6&7 ? I think I damaged the chip contact, the image had interference, I removed the picoboot and redid the chip solder, so it normalized, but I'm afraid to solder there again.

EMaDeLoC, 2022-07-06:
I have no feed problem, too. But it's different.
I have GC with faulty drive. I installed PicoBoot and if I turn on without the drive, black screen. Turn off, I place DVD drive, turn on, PicoBoot is working. Turn off, DVD drive out, turn on, PicoBoot still working.

I read about problem with long wires, but mine works, but I have problems without DVD drive installed.

BTW, GC PAL DOL-001

webhdx, 2022-07-06:
@EMaDeLoC keep wires as short as possible. I'm running my test cube without the drive and don't have any problems. I'm pretty sure your wiring need rework.

webhdx, 2022-07-06:
I've tested it with BBA personally and there are no issues. Check your wiring. You guys need to start posting photos of your setups so we can find what they have in common.

DacoTaco, 2022-07-06:
> I have tested mine at several stages of re-assembly with 100% boot success right up until I inserted the broadband adapter. With the broadband adapter installed I get a black screen, without the broadband it boots as expected. This is reproduced with both sd2sp2 and a memory card sd adapter guy.
> 
> Is this known or expected?

ive had the same with my setup but all issues got resolved when i shortened my wires. 
also, make sure you bridged Pin1 on the serial connector 1 with ground correctly. besides using that bus, the BBA semi shorts those pins itself

maniaco007, 2022-07-07:
Just passing by to let you know that everything worked out!
I redid the welds very carefully!
Now just enjoy!

Yuuki2012, 2022-07-07:
Same for me. Everything worked when using thiccer and shorter wires. Used about 15-20cm wires. And it instantly booted to Swiss.

webhdx, 2022-07-07:
Cool. This is the most common issue people are experiencing right now. You need to keep wires as short as possible + use thicker wires for 3.3V and GND. Since OP @Yuuki2012 fixed his setup I will close this issue.

closed 2022-07-07 by webhdx