webhdx/PicoBoot #13: Raspberry Pi Pico W compatibility

Hackinside, 2022-07-07:
| Answer some general questions first                             	|  	|
|:-------------------------------------------------------------------	|:------:	|
| Are you my GitHub sponsor? | no |
| Did you check older issues to see if it was requested before? | yes|

Raspberry Pi Pico W has now Status LED addressed to GPIO0  instead GPIO25

webhdx, 2022-07-08:
Hey @Hackinside 

I received my Pico W a few days ago but haven't started working on it yet. The board should be compatible even without needing to recompile the firmware, it's just the LED that doesn't work at the moment.
I plan to add some PIco W specific features like OTA updates. But for that I'd rather wait until someone from Raspberry Pi makes an official solution so I don't have to code it myself.

Hackinside, 2022-07-08:
Thank you for your reply @webhdx quick reply. 

That is a great idea to add an OTA update. Also, I was wondering if ( PicoBoot W ) will be capable to unlock the game cube network capabilities ( Emulate the broadband adapter DOL-015 )

webhdx, 2022-07-08:
I haven't looked into it but Pico W is not as powerful as people may think. I definitely want to experiment a little bit and see what can be done but getting it act as 1:1 BBA replacement is very unlikely. Maybe I can get it to work with some homebrew apps like Swiss to synchronize time via NTP, download updates etc.

JamesAZ0, 2022-07-09:
Just to chip in, I can confirm first hand that the current package works fine on the Pico W. Installed without issues tonight.

nex86, 2022-07-09:
> Thank you for your reply @webhdx quick reply.
> 
> That is a great idea to add an OTA update. Also, I was wondering if ( PicoBoot W ) will be capable to unlock the game cube network capabilities ( Emulate the broadband adapter DOL-015 )

I doubt it. the wifi would probably not fast enough for game loading.

webhdx, 2022-07-09:
I've updated FAQ with Pico W question: https://github.com/webhdx/PicoBoot#can-i-use-new-pico-w-board

I'll close this issue for now.

closed 2022-07-09 by webhdx