webhdx/PicoBoot #14: Bypass Pico on boot

rfancella, 2022-07-08:
| Answer some general questions first                             	|  	|
|:-------------------------------------------------------------------	|:------:	|
| Are you my GitHub sponsor? | no 	|
| Did you check older issues to see if it was requested before? | yes 	|

I think this is a great project!  This will save many a DVD drives out there.
Could a feature be added to bypass and boot normally with a specific button being held down?  Possibly the Z button??

webhdx, 2022-07-08:
It's already possible. Do not put one of x/y/z.dol files and hold matching button to load default menu.

kimata2001, 2022-07-09:
> It's already possible. Do not put one of x/y/z.dol files and hold matching button to load default menu.

Dear webhdx

Thank you for your amazing work

however, I couldn't find any x.dol y.dol z.dol.
I looked your released and code and SWISS release also.

I so sorry to bother you but could you offer more description about this bypass booting feature and x/y/z.dol files?

webhdx, 2022-07-09:
You didn't understand. You can rename **any** app to `x.dol`, `y.dol`, `b.dol` and run it by holding down X/Y/B button during boot. If you don't put one of these files but still hold designated button during boot it will load default GC menu. This is a fallback behavior when the file can't be found on an SD card.

To put it simple:
- make sure there is no `b.dol` file on your SD card
- hold B button during boot
- enjoy default boot animation and menu

kimata2001, 2022-07-09:
> You didn't understand. You can rename **any** app to `x.dol`, `y.dol`, `b.dol` and run it by holding down X/Y/B button during boot. If you don't put one of these files but still hold designated [...]

OH! Thank you so much for the fast answer!
I got it now 
Thank you so much

webhdx, 2022-07-09:
No problem. Enjoy PicoBoot! I'll close the issue but feel free to open a new one if you have more questions.

closed 2022-07-09 by webhdx