webhdx/PicoBoot #15: GC wont "boot" with picoboot powered up.

requestedeverywhere, 2022-07-09:
| Answer some general questions first                             	|  	|
|:-------------------------------------------------------------------	|:------:	|
| Which firmware version are you using?                             	| `v0.1` 	|
| Did you compile it yourself?                             	| no 	|
| Do you use alternative install method not mentioned in the guide? 	| no 	|
| Are you using official Raspberry Pi Pico board?                   	| yes |
| Did you pay someone to install it for you?                        	| no |

I triple checked the solder joints. and the pinouts to make sure I had done it all right. and to be clear i think the pico is doing what its supposed to just not all the way. with the pico powered up the gc hangs on boot. with the power for the pico disconnected (just the 3v3) the gc boots normally. I tested removing the wires one at a time to see which one would make it "fail" to boot. but it only does is when they are all hooked up. the board boots with the gnd un-soldered  and with it hooked up and bridging the 2 connections as instructed. I also tested with the individual pins unhooked then hooked back up in order including just having gnd unhooked but 3v hooked up and it booted happily every time. (definitely not dealing with a bad solder joint or a bricked GC here) to be clear i believe the mod is doing as it should and interrupting the normal boot, but afterwards nothing happens.  any clues as to what I might be doing wrong? I have an sd2sp2 and my swiss dol in place as instructed.

nex86, 2022-07-09:
swiss dol needs to be named IPL.DOL
but even if no matching DOL is found it should still boot the vanilla gamecube bios.

webhdx, 2022-07-09:
What does it mean "hangs on boot"? If you get black screen then it's for sure bad wiring - can be too long or too thin wires. If it freezes or glitches out then it's a problem with the console and not the modchip.

nex86, 2022-07-09:
What cable length is fine? I might switch to 30cm cables because I want to mount the pico on the rear of the case and cut a whole in it for the usb connector

webhdx, 2022-07-09:
30cm is way too much. 10-15cm is preferred length. Just mount it on the fan assembly as shown in RockerGaming video. You don't need to cut any holes for USB connector because to flash the chip you still need to access BOOTSEL button. I don't think ruining your case is worth it, especially when there are only 4 screws to undo in order to flash the chip. And next update will include Pico W support which you may want to replace your PicoBoot with in order to get WiFi updates.

webhdx, 2022-07-09:
BTW: https://github.com/webhdx/PicoBoot#my-console-doesnt-start---i-see-a-black-screen

requestedeverywhere, 2022-07-09:
> swiss dol needs to be named IPL.DOL but even if no matching DOL is found it should still boot the vanilla gamecube bios. Make sure you didn't bridge anything.

bridges are all checked for,only ones present are the ground and gp6 gp7.

> swiss dol needs to be named IPL.DOL but even if no matching DOL is found it should still boot the vanilla gamecube bios. Make sure you didn't bridge anything.

tried multiple times on this, even made sure to format with windows just in case (im a linux user), also by hangs at boot im not getting a black screen, im getting no screen. (gamecube comes on, scaler doesnt see any video signal) that said i am using 24Ga wire. its possible i made the wires too long / thin and thats causing this. I will check that shortly.

requestedeverywhere, 2022-07-09:
also noticed the what if my screen is black error on the readme (didnt see that there yesterday) and as i said above I will be checking my wire length. i had planned to put the pi on the front like macho nacho did. but as its not working that may change.

webhdx, 2022-07-09:
I'm almost sure you are using too long wires.

requestedeverywhere, 2022-07-09:
> I'm almost sure you are using too long wires.

we will find out for sure today. thinking about it, i get why that could be an issue. and it shouldnt be too hard to test and fix. thank you for the time though, im sure you are getting banged over the head with all the hype around this mod, and im thankful you took the second to address this, even if it is just a rookie mistake.

KazeEternal, 2022-07-09:
18CM with 30awg wire. Sounds like I need 26awg, the lenght maybe too long? I'm experiencing the same issues.

webhdx, 2022-07-09:
Try using thicker wire for GND and 3.3V

nex86, 2022-07-09:
I'm currently using 15CM jumper wires what people use for breadboards and stuff.

requestedeverywhere, 2022-07-09:
> I'm almost sure you are using too long wires.

as promised i tested that, and you where right. I used 24ga wire when i did it and no change to the wire gauge was needed all i had to do is shorten ALL the wires and it boots. i tested just shortening the gp connections first and it tried to work once then didnt do it again shortened the gnd and tried no boot, then shortened the power and shes working reliably.

jbarker2160, 2022-07-09:
Are the internal pull-up/pull-down resistors in place? I've used i2c and spi over 33cm 30ga wire without issue on the pico when using the internal resistors.

KazeEternal, 2022-07-09:
30awg wire, shortened the lengths. works fine.

Had some cruft on the board that I found after removing the mod. Re-applied once I fixed the problem. I don't think the wire gauge is the problem. 

anyway I'm up and running. Glad to hear requestedeverywhere is good to go as well.

webhdx, 2022-07-09:
It's a common issue it seems. Glad it works for you now.