webhdx/PicoBoot #16: IGR to Swiss Menu

maniaco007, 2022-07-10:
Has anyone managed to get the IGR working? (In Game Reset)
Mine when it restarts, it goes back to the original boot screen, I need to turn off and on the console to access Swiss.

webhdx, 2022-07-10:
Make sure you are using `igr.dol` setting. It's actually quite buggy and doesn't work all the time. I hope to fix this issue in the future but for now `igr.dol` is your best option.

maniaco007, 2022-07-10:
Yes I am using igr.dol.
In all the games I tested, it restarts the console, so far so good.
However, it does not return to the swiss screen, it accesses the main menu of the console.

webhdx, 2022-07-10:
This is an issue with both Swiss and PicoBoot. I can't offer you any solution at the moment but I'm fully aware of it happening and I'll try to address it in future updates.

maniaco007, 2022-07-10:
Thanks for the answer!

And congratulations, your project is fantastic!
I'm from Brazil, here it's been very successful!

Extrems, 2022-07-10:
Make sure you aren't using exFAT. The reload stub currently doesn't support it.

maniaco007, 2022-07-10:
Using fat32

Extrems, 2022-07-10:
You do have a `ipl.dol` or `igr.dol` file in the root directory of your SD card, right?

maniaco007, 2022-07-10:
> Você tem um ou arquivo no diretório raiz do seu cartão SD, certo?`ipl.dol``igr.dol`

Yes, both.
I changed the conf. in swis to igr.dol, but it restarts to the console boot screen, it doesn't come back through swiss, only when I turn off and on the console. P.s. using picoboot.

Extrems, 2022-07-10:
It doesn't matter that you're using PicoBoot. A restart means the reload stub failed to load a DOL file from any SD card in the system.

maniaco007, 2022-07-10:
[image]

dephcon, 2022-07-13:
I'm having the exact same issue.  SD card is formatted exFAT, ipl.dol and igr.dol in the root with latest version of swiss.

When I configure "In-Game Reset" to "igr.dol" it takes me back to the stock gamecube boot screen, which is the same behavior as setting it to "Reboot"

Extrems, 2022-07-13:
Again, exFAT is not supported by Swiss' reload stub.

dephcon, 2022-07-13:
@Extrems my apologies, I miss read your comment as you MUST use exFAT.   I have a larger SD card and require exFAT to format it in Windows.  I think I need a 3rd party utility to force fat32, I'll look into it and it might be worth mentioning on the swiss docs :)

maniaco007, 2022-07-13:
> @Extrems minhas desculpas, eu li seu comentário como você deve usar exFAT.
> 
> Usei [fat32format](http://ridgecrop.co.uk/index.htm?guiformat.htm) para reformar meu cartão de exFAT para FAT32 e está tudo bem!

after formatting, is the reset working?

maniaco007, 2022-07-13:
That was the solution to the problem!
I formatted the card again with fat32format, now it works perfectly In Game Reset!

Thank you to everyone who somehow contributed.

closed 2022-07-13 by maniaco007