webhdx/PicoBoot #21: Boot original GameCube IPL after install PicoBoot

giancarlozero, 2022-07-21:
Hello. I have a little question.

I'm new to GameCube modding and I want to install PicoBoot on my console.

After installing PicoBoot, will I be able to boot the original GameCube IPL if I want to? Or will PicoBoot always boot Swiss and never access the original IPL again?

Thanks.

(Sorry about my poor english. Not my native language)

closed 2022-07-24 by webhdx

giancarlozero, 2022-07-27:
> Hold one of X/Y/B during boot and make sure there is no matching x/y/b.dol file on your SD card.

Excelent! So I will mod my GameCube with PicoBoot for sure.

Thank you and congratulations for this awesome project. =)

FibreFoX, 2022-08-03:
@webhdx Can you add this to the FAQ-list please? Probably is very helpful to know that this is still accessible (an is not yet mentioned anywhere than here).

webhdx, 2022-08-03:
This feature comes from https://github.com/redolution/iplboot and I don't want to document iplboot features here in PicoBoot repository.

FibreFoX, 2022-08-03:
I understand that, but coming from a "I dont know the technical details" perspective, I would never look over there ;) Maybe some hint like "for usage, please look over at that project" would help others to find this.