webhdx/PicoBoot #24: Flat Battery

Rtrobits, 2022-07-30:
| Answer some general questions first                             	|  	|
|:-------------------------------------------------------------------	|:------:	|
| Which firmware version are you using?                             	| `v0.1` 	|
| Did you compile it yourself?                             	| yes/no 	|
| Do you use alternative install method not mentioned in the guide? 	| yes/no 	|
| Are you using official Raspberry Pi Pico board?                   	| yes/no 	|
| Did you pay someone to install it for you?                        	| yes/no 	|

If the battery goes flat (or you change battery) and the gamecube console loses the date and time, the console wont boot with picoboot installed until you power on with the BOOTSEL button being pressed. Once the console powers on you can reboot the console and picoboot will launch.

kahardin, 2022-07-30:
Thank you for posting this - I have been pulling my hair out thinking I did something wrong with the wiring. Would be nice to see this covered in the "My console doesn't start - I see a black screen." section of the README.md as it presents itself the exact same way. I tore my system apart shortening wires and checking continuity for probably a cumulative 12 hours before coming here after giving up.

webhdx, 2022-07-31:
I couldn't reproduce the issue. Something tells me your wiring may not be optimal.

Rtrobits, 2022-08-01:
All wires were around 10cm length and were AWG 28 as advised. I've now ordered a different batch of AWG 28 wire to see if that is the issue. I'll post back once received and installed with new wire.

kahardin, 2022-08-01:
I also used 28 AWG, I intend to switch to 26 AWG as I've seen scattered reports of inconsistent booting with 28 AWG. My current wires are between 5 and 6 inches in length. Waiting on the 26 AWG to arrive.

webhdx, 2022-08-01:
Don't replace the wires if it's working for you. No need to stress the motherboard and components with excessive heat. Act only if experience any booting issues (no video).

Rtrobits, 2022-08-01:
I mod consoles to sell, so the last thing I need is customers having issues so I've removed the wires and I'm now awaiting a new delivery of new ribbon cable.
I also always change the battery to a socket and new battery on all NGC consoles - hence how I got the problem in the first place.

I've seen that the 3.3v is recommended to be AWG 26. As I used AWG 28 could that be the issue? Is AWG26 a recommendation or a requisite?

webhdx, 2022-08-01:
> I've seen that the 3.3v is recommended to be AWG 26. As I used AWG 28 could that be the issue? Is AWG26 a recommendation or a requisite?

As with any electronics you want thicker wires everywhere where more current passes through. It is especially important to have solid ground connection.

webhdx, 2022-08-01:
BTW. I have a rule to not provide any help to people who take money for PicoBoot installation. In my understanding you'd be qualified enough to know all that stuff you're asking about.

closed 2022-08-01 by webhdx

nicman23, 2022-08-01:
i actually have a flat battery and cannot go to the menu - no dvd drive. it boots straight to swiss normally.

webhdx, 2022-08-01:
Yes, this is why I suspect wiring issue in @Rtrobits case.

Rtrobits, 2022-08-02:
> 

I fully understand and accept that rule, However as this is any new device (and can I say I think what you have achieved and released is fantastic) I only posted to let others know of my situation and the solution I found as I'm sure there would be others out there that would be in the same spot as myself.
I wanted to know the cause as I had used the recommended gage wire (with the exception of the 3.3v - but have seen others working on ALL AWG 28 only too) so assumed the AWG 26 for 3.3v was just a precautionary measure to build in more tolerance - clearly it's not lol

I've still not received my new wire and will post back when I have. It's looking like not all AWG 28 wire is equal

webhdx, 2022-08-02:
You can also try different thing - solder additional GND wire and bundle the wires together so they run close to each other. Put them in a heatshrink tubing.

kahardin, 2022-08-20:
Posting an update for the curious as I keep seeing these flat battery issues referenced in other places such as Reddit:

I have three different DOL-001 cubes available to me, and I finally got some more pico boards in (was waiting for the wireless variants to do a bit of future-proofing in case wireless features are ever supported) and ran through the install process on a fresh unmoved GC:
- I used 26 AWG instead of 28 AWG, and the wires were longer (maybe too long, I had a hard time putting the case back together) 
- I installed without removing the battery
- Booted fine
- I removed the battery
- I replaced the battery with a socket and put a battery in
- Booted fine
- Let it run for awhile, played GBI, played an actual disc (REmake)
- It has yet to fail a boot

So I put that one out as the one I use to play on, and grabbed the inconsistent one that had this issue. Replaced all the wires with 26 AWG, took the battery out for good measure and it booted fine. Played GBI, booted up REMake via the DVD drive… It also hasn’t failed a boot since swapping the wires.

In my case, it seems that the switch to quality 26 AWG wires that helped me, and I hope this info is helpful to someone else if they have this problem. Don’t cheap out on the wiring!