webhdx/PicoBoot #25: Just an install question

rayrayrayraydog, 2022-07-31:
| Answer some general questions first                             	|  	|
|:-------------------------------------------------------------------	|:------:	|
| Are you my GitHub sponsor? | no 	|
| Did you check older issues to see if it was requested before? | yes 	|

Do I *need* to reinstall the optical drive with Picoboot for any reason?  Mine is broken and I would rather leave it out.

webhdx, 2022-07-31:
Vanilla IPL will display error message if there is no drive installed. Swiss can work without the disc drive. If you don't plan to use standard GC menu then you should be fine with no disc drive.

rayrayrayraydog, 2022-07-31:
I'm new to 'cube modding.  I assume Swiss is all I need.  Thanks for clarifying.  Leaving out the optical drive should also keep the Pico more easily accessible for updates.

closed 2022-07-31 by rayrayrayraydog

Extrems, 2022-07-31:
I recommend leaving it in.

rayrayrayraydog, 2022-07-31:
What are the advantages of leaving it in then, if it doesn't work?

Extrems, 2022-07-31:
You can expect a lot of software to hang for at least 10 seconds if there's no disc drive present.

rayrayrayraydog, 2022-07-31:
I see.  I guess I'll try it both ways and see what happens.  Thanks.