webhdx/PicoBoot #27: Select Boot Image

maniaco007, 2022-08-02:
It would be possible to create a way to customize the boot screen, just like this user did: https://youtu.be/sCODnY4vml4 .

Unfortunately he did not explain how to proceed.

kjuno, 2022-08-03:
This approach requires modifying and recompiling iplboot, see https://github.com/redolution/iplboot for compiling for PicoBoot

webhdx, 2022-08-03:
Yes you need to compile your own version of iplboot (or whatever you want). We are working on getting patched IPL to run from PicoBoot, this will bring back cube animation before Swiss loads.

kjuno, 2022-08-03:
wouldnt be possible to use a video file loaded from sd for the startup sequence?

webhdx, 2022-08-05:
Anything is possible but you have to code it yourself. Iplboot is meant to be minimal and it's unlikely anything like this will be added to it anytime soon. If there is interest in it maybe you should fork iplboot and try adding it yourself.

closed 2022-08-05 by webhdx