webhdx/PicoBoot #28: Possible to use WiFi on a Pico W board?

5p4rkP1ug, 2022-08-04:
Would it be possible to tap into the serial interface that an Ethernet adapter would use, and emulate it on a Pico W and have WiFi on your GameCube?

webhdx, 2022-08-04:
I get asked this a lot. I haven't tried anything with Pico W's WiFi yet and I don't know how good performance we can achieve. I highly doubt it'll be useful for streaming games or emulating LAN connection for Mario Kart. Some light network stuff for homebrew like providing updates to Swiss or making Homebrew Browser could work though.

5p4rkP1ug, 2022-08-04:
Light network stuff is good. I want to manage files on my SD card remotely, either by a terminal or FTP service (SSH would be perfect for this).

webhdx, 2022-08-05:
I'll let myself close this issue as it's been answered already. Feel free to open another issue if you have more ideas for PicoBoot.

closed 2022-08-05 by webhdx