webhdx/PicoBoot #29: picoboot does not work with some power cables

azareth012, 2022-08-04:
I have installed the raspberry pi pico with picoboot correctly, when turning on, picoboot sometimes does not work and the original gamecube menu appears, the console works well with discs and everything seems to be fine, I tried another power cable and now picoboot always works.

This confirms that some power cables have defects over time, they work fine with a "virgin" console but not with picoboot.

What component of the power cable may be defective so that picoboot does not boot sometimes? a capacitor maybe? this is important, because it can be the reason that many installs of picoboot don't work and people think it's because they installed it incorrectly.

My console is a DOL001 and I use SD GECKO, thanks.

webhdx, 2022-08-04:
I assume you meant power supply by "power cable"? It's hard to tell what was the issue but posting your PicoBoot install photos would tell me more. I suspect this is another example of unoptimal wiring.

azareth012, 2022-08-04:
> I assume you meant power supply by "power cable"? It's hard to tell what was the issue but posting your PicoBoot install photos would tell me more. I suspect this is another example of unoptimal wiring.

yes, I mean the power supply, with another power supply it works correctly 100% of the time, I forgot to take pictures of my wiring inside, but I'm sure it's correct, the length of the cables is short, there is continuity at all and there are no cold welds.

With one power supply picoboot works sometimes and with another power supply it always works so it is sure that it is the power supply, right? Thank you

https://ibb.co/8mJNNQS

webhdx, 2022-08-05:
It's hard to tell without seeing how you soldered PicoBoot. If replacing power supply fixed your issue I don't think there is anything I can to make sure it doesn't happen in the future for other people. This is strictly related to the hardware.

closed 2022-08-05 by webhdx

azareth012, 2022-08-05:
I think it's important to at least mention it in the FAQ, "if you're sure your installation is good and picoboot isn't working properly, try another power supply" maybe there are people out there trying to fix their problem by re-soldering over and over again and end up damaging your console.

I did enough tests with 2 SD Gecko, 3 micro SD, 2 raspberry pi pico, cutting cables and re-soldering, none of this solved my problem, when I changed the power supply the problem was completely solved.