webhdx/PicoBoot #30: Flat Battery (revised)

Rtrobits, 2022-08-05:
After the first issue was closed without me being able to advise if AWG 26 wire on 3.3v and GND fixed the issue (which it hasn't) and knowing other people are reporting as having the same issue I have had to make another report.
Since my last report, I have used different AWG28 wire for the data connections and now AWG26 for 3.3v & GND as suggested by yourself.
Wiring lengths are also well within the guidelines quoted for optimal installation.
I have tested picoboot on 10× NTSC consoles. Of these, 3 of them experience the same issue of not being able to boot if the date and time are not set and the battery is flat, however changing the battery for a known good battery and then turning the gamecube on with BOOTSEL pressed allows the date and time to be set. Once this has been done no issues are experienced. 

It may be a coincidence at this point, but all 3 that have issues are early revision NTSC DOL-CPU-10 or DOL-CPU-11 motherboards (The ones with the pink 'PRESS START' screen. I believe these are the few consoles that use the uncommon NTSC IPL Rev 1.0.

If you require any more info, please let me know.

[Install wire length](https://ibb.co/xs8RVr2)

webhdx, 2022-08-05:
I'd be very surprised to learn it's caused by the battery because PicoBoot replaces whole BS2 where battery/date check is performed. I think this is just a coincidence. It's hard to judge your soldering from the picture but ground wire definitely needs rework - it looks like you haven't fully melted the solder. GND usually needs much more heat due to large dissipation area.

Rtrobits, 2022-08-05:
** UPDATE ** I got Picoboot running fine, resetting, turning off, then on again and no problems. Left plugged in (but tuned off) for 15 minutes and console has returned back to requiring BOOTSEL to be pressed at power on even tho date and time are still saved.... so issue may be related to something else.

Rtrobits, 2022-08-05:
[image]

Rearranged the wires to give a better view. The GND wire is definitely connected as both points were flowing.

webhdx, 2022-08-05:
Looks good. No idea why this happens. I couldn't reproduce it on my consoles. Let's see if more people reports the same issue in the future.

Rtrobits, 2022-08-05:
No problems. 

I'll keep plugging away at it and let you know if I get it resolved - probably something simple that I've overlooked somewhere else especially If you cant replicate with the same version board then I'm assuming it's the boards I have.

I'll remove this pico and give the mobo another full clean with IPA and test again, then may look at possibly doing a recap to see if that helps

Thanks once again.

closed 2022-08-05 by Rtrobits