webhdx/PicoBoot #31: Update Question

GabyPCgeeK, 2022-08-07:
In the "Updating PicoBoot" section it says to desolder the 3.3V pin but,

Has anyone tried to updating the rpi pico by grounding 3V3_EN, plugging the usb, holding down BOOTSEL and turning on the console?

webhdx, 2022-08-07:
That could also work. You can power on the console with BOOTSEL button pressed down and program it  without (de)soldering anything.

GabyPCgeeK, 2022-08-07:
Tested it successfully. The pico only powers on when the console is on. Left the console on for a some time while connected to usb and there no component was heating up. When powering on the console while holding the BOOTSEL button it boots normally.

closed 2022-08-07 by GabyPCgeeK

shawnwall, 2022-08-13:
@webhdx @GabyPCgeeK if I'm reading the comments above correctly, it sounds like there _is_ a safe way to update pico without desoldering? I just wanted to clarify. If that's true, we should update the content on https://github.com/webhdx/PicoBoot/wiki/Updating-PicoBoot