webhdx/PicoBoot #32: Picoboot testing problems with power supply

azareth012, 2022-08-09:
Hello, a few days ago I opened a topic commenting that some power supplies do not work with picoboot, I bought a third power supply to do more tests, these are my results.

Three power supplies in total, two american power supplies and one japanese power supply, which we will call 1PUSA(red), 2PUSA(green) and 3PJAP(green) from left to right as seen in the photo, I have checked the voltage and amperage of the 3 and the values ​​are correct and I used 3 different consoles for these tests, one with picoboot USA, another with GC loader USA and one not modified JAP.

1PUSA:
-Works with a console without modifying JAP
-Works with a console with GC Loader USA
-It does not work with a console with picoboot USA, when turning on the console sometimes it does not start swiss and it sends directly to the original menu of the gamecube, as if there was no SD memory inserted.

2PUSA:
-Works well with all 3 consoles.

3PJAP:
-Works well with all 3 consoles.

The console with picoboot USA is a DOL001 with digital and analog video output, but it does not have serial port 2, so all the tests were done through port B of the memory with SD Gecko, picoboot is installed correctly with cables short, AGW28, and with continuity in all its points, it is a clean and functional installation (that's why it works well with the other 2 power supplies).

Try to get 1PUSA to work by all possible means with picoboot, use 2 different SD Gecko, 6 SD to micro SD adapters, 3 micro SD of different brands and capacities, 2 raspberry pi pico reinstalling picoboot on each one, I have changed the wiring completely I have rechecked the welds.

1PUSA is still intermittent with picoboot while the other 2 work fine with picoboot on the same test console.

This is a proof that picoboot can have problems with some power supplies, I consider this very important and I ask the author of picoboot to consider putting this in the FAQ, some people have problems with picoboot and are changing wiring and re-soldering over and over again because that's what they are told to do, even break their consoles.

There should be a warning that says "If you think your installation is correct and picoboot isn't working right, try another power supply" or something like that, power supplies can currently be purchased for 5 bucks and that can prevent someone from breaking your console.

I can donate the power supply with problems if someone is interested, I'm from Mexico.

[image]

webhdx, 2022-08-09:
Thanks for a very detailed explanation. I was contacted by another person who has very similar issue - they get succesfull PicoBoot boot only if they use analog output. For some reason it loads vanilla menu if they use digital adapter. I'll update troubleshooting guide with for findings. Don't worry. And thanks again for your dedication in finding it out.

azareth012, 2022-08-09:
> Thanks for a very detailed explanation. I was contacted by another person who has very similar issue - they get succesfull PicoBoot boot only if they use analog output. For some reason it loads [...]

Thank you, feel good to be able to help, i dont have an analog video cable, i broke it a few years ago, i currently use a bitfunx hdmi adapter on the digital output, i will get an analog video cable to try, the hdmi adapters for the digital output have been become popular and I'm sure that console fans at least have one of these, so this problem is now more important.

azareth012, 2022-08-09:
I forgot to comment, in all 3 consoles I use the digital video output, so for some reason the 1PUSA power supply is still the problem, since with the other power supplies it works correctly.

azareth012, 2022-08-09:
I forgot to comment, in all 3 consoles I use the digital video output, so for some reason the 1PUSA power supply is still the problem, with the other power supplies it works correctly, my console with GC Loader has an SD2SP2 adapter of the so-called "Game Boy Player compatible" and the console with picoboot uses a SD Gecko WiiSD, even so when I use the P1USA power supply and the digital video output, the console sometimes starts In the original menu of the console, I insist that the power supply is the problem.

webhdx, 2022-08-16:
I've updated the Troubleshooting guide and linked this issue so people can get more information here.

closed 2022-08-16 by webhdx

VirusABC, 2022-08-17:
Thanks for your research! 

I've been trying to install picoboot on my GC for over than a week and always had the blank screen... at first I agree that it was due to my soldering, but I think that I did enough tries that should work at least once... Tried stranded wires, solid wires, 24AWG, 26AWG, cooper, aluminun, short wires, passing the wires close to each other, same lenght wires, picoboot v0.1, v0.2... There's just one more thing to try for sure: My rasp pico itself... I've tested it by flashing the micropython and it seemed to work... I'll try to flash the pico controller and turn it into a joystick. If it doesn't work, I'll report and try to get another one.

My GC is using a PSU similar to the one in the middle, except that it is written "MADE IN JAPAN" instead of "MADE IN CHINA". I believe that It is one of the first batch of US GC DOL-001 models, made in Japan for US launch. Under the heatsink, it reads DOL-CPU-10 and PCB is written "Made in Japan" just under the drive connector.

Unfortunately I don't have any friend that has a GC nearby so I couldn't test with another PSU. And a GC PSU around here is not so cheap and not guaranteed to work with pico :(

The picoboot would help me replacing my failing disc drive (it takes 15 minutes to "heat up" and start reading original discs). I guess that I'll try to replace the capacitors on the drive's board and stick with the exploited game save on the memcard for now. Thanks for your time and for including this on the FAQ.

boldandbrad, 2022-09-18:
@azareth012 is there a difference in model/revision of the two USA PSUs you tested? After reading this thread I'm wondering if my PSU is causing intermittent boots to native bios instead of swiss. Just wondering if there is a specific PSU version I should track down or if it doesn't matter.

azareth012, 2022-09-18:
> @azareth012 is there a difference in model/revision of the two USA PSUs you tested? It's hard to read the far left one in the image.
> 
> After reading this thread I'm wondering if my PSU is causing intermittent boots to native bios instead of swiss. Just wondering if there is a specific PSU version I should track down or if it doesn't matter.

internally there are some different components, today I have tried 4 usa PSU and 1 jap PSU, the problem persists only with 1 of the usa PSU, I have tried it on 3 gamecubes in all there are problems with picoboot

boldandbrad, 2022-09-18:
> internally there are some different components, today I have tried 4 usa PSU and 1 jap PSU, the problem persists only with 1 of the usa PSU, I have tried it on 3 gamecubes in all there are problems with picoboot

Thanks for the reply! After I posted above I remembered I had a box of random GC stuff in a closet that I haven't touched in awhile. Turns out I had a total of 4 PSUs (3x USA Made in China and 1x USA Made in Japan). I had the same issues with all 4.

However, I did solve the problem. It was unrelated to the PSUs but rather the fault of my Eon GCHD Mk 2. I may post a separate issue where I explain my findings in depth like you did here so webhdx and others may learn from them.

Thanks again! Your comments helped me narrow down the issue.