webhdx/PicoBoot #33: Format your SD card to FAT32 or exFAT. Download the latest Swiss release

alimao10, 2022-08-13:
| Answer some general questions first                             	|  	|
|:-------------------------------------------------------------------	|:------:	|
| Are you my GitHub sponsor? | yes/no 	|
| Did you check older issues to see if it was requested before? | yes/no 	|

. Preparing SD card

Format your SD card to FAT32 or exFAT. Download the latest Swiss release from [here](https://github.com/emukidid/swiss-gc/releases/latest) and grab swiss_rXXXX.dol file, rename it to ipl.dol and copy to the root of your SD card.

swiss does not recognize exfat formatted card I think there is an error there

Extrems, 2022-08-13:
Swiss does recognize exFAT and GPT.

alimao10, 2022-08-15:
> Swiss does recognize exFAT and GPT.

tranks I tested exFAT it and it worked I didn't know thanks

last doubt on the wii . swiss doesn't work . get a black screen

webhdx, 2022-08-16:
Swiss is compatible with exFAT: https://www.gc-forever.com/wiki/index.php?title=Swiss#Supported_Devices

I don't think PicoBoot repository is a good place to report issues with Swiss running on Wii. I consider initial question as answered.

closed 2022-08-16 by webhdx