webhdx/PicoBoot #36: Possible code for Seeed Studio XIAO RP2040

caprisun08, 2022-08-26:
| Answer some general questions first                             	|  	|
|:-------------------------------------------------------------------	|:------:	|
| Are you my GitHub sponsor? | yes/no 	| no
| Did you check older issues to see if it was requested before? | yes/no 	| yes

I know the pi pico is cheaper, but this is smaller I am not sure if the pinouts are the same type but I was wondering if the code was useable on [this](https://www.amazon.com/dp/B09NNVNW7M/?coliid=I2YWC7D1LD2NZ3&colid=15Y3POTJ7DTCA&ref_=lv_ov_lig_dp_it&th=1) or can be tweaked?

webhdx, 2022-08-26:
Unfortunately I only support the official Raspberry Pi Pico board. If you want to use any other board, I won't be able to help you. It's explained in the wiki.

closed 2022-08-26 by webhdx

caprisun08, 2022-08-26:
sorry about that, ok!