webhdx/PicoBoot #37: Gamecube powers on black Screen

Franken83, 2022-09-01:
| Answer some general questions first                             	|  	|
|:-------------------------------------------------------------------	|:------:	|
| Which firmware version are you using?                             	| `v0.2` 	|
| Did you compile it yourself?                             	| no 	|
| Do you use alternative install method not mentioned in the guide? 	| no 	|
| Are you using official Raspberry Pi Pico board?                   	| yes 	|
| Did you pay someone to install it for you?                        	| no 	|

Gamecube powers on 
Power LED on
Pico LED on
Fan ON
Screen Black

BUT If I press Bootsel on Pico and power GC on it will load regular GC Boot and play game discs

Wires are 2inches long,  28awg for GP4 GP5 and GP6&7  
All Solder Points are clean (all continuity tested)

So I have read all the posts about black screen but none specific to this however I did try all the "tricks" 
-Shortened Wire
-Heavier Gauge for power wires (3.3v, GRND)
- I also switched to Picoboot v0.1 just to see but nothing

I haven't tried changing the power supply but since the Pico has power could that even the issue?

This doesn't make sense to me 

PLS HELP LOL

webhdx, 2022-09-01:
Everything seems to be correct according to what you wrote. Black screen would indicate hardware issue and that can be only validated with installation pics.

luciolles, 2022-09-03:
Hello there, I do have the exact same problem than @Franken83. I thought at the begining I had fried up my GC. I unmounted all the mod installation. Boot up the GC and it was all good. I resoldered all of it. Back Screen of death again.

I did try the little hack of presing Bootsel on Pico and power GC on and it will load regular GC Boot and play game discs too.

Hum.. Kinda lost here 😭

Might post some close up pictures later on

webhdx, 2022-09-03:
Thanks @luciolles. Glad it works for you now. These kind of issues are almost always caused by wires being too long which causes interference. I'm still looking into a way how to optimize the wiring but ultimately the best solution will be using a custom made flex cable.

Zonark, 2022-09-04:
Just had a quick question are you holding the boot select button down the entire time when you plug it in to the PC and then drop the firmware in there? Also, is the LED lighting up on the pico itself? Can you upload any photos of your install on the pico board side?

ThexXTURBOXx, 2022-09-13:
Something I might want to point out in that regard:
I also had the same problem at first and solved it by using shorter wires.
However, this page:
https://github.com/webhdx/PicoBoot/wiki/Troubleshooting-tips#orange-light-is-on-but-it-doesnt-display-anything-on-the-screen
says that 10 cm is about maximum length.
This contradicts the comment
https://github.com/webhdx/PicoBoot/issues/15#issuecomment-1179537902
where 15 cm is also acceptable. In my experience, 15 cm is perfectly fine, but anything beyond ~17-18 cm seems to cause problems.
Also, I was not able to find a good way to wire the Pico with only 10 cm of wires. It seems like 13 cm might be minimum length to place the Pico behind the small "heat sink window" of the GameCube or however you might want to call this :)
[image]

webhdx, 2022-09-13:
@ThexXTURBOXx It also greatly depends on the quality of the wires used. 10cm is perfectly enough for the install and this is what I recommend now.

@luciolles This location is not recommended. Even if it works for you people shouldn't use that spot. In the future PicoBoot will be updatable through the application running on GameCube. There will be no need to physically access the modchip.

Skullator, 2022-09-13:
Regarding updating through app with no need to physically access the pico. Would this be true only if a wireless networking capable pico is used?

webhdx, 2022-09-13:
No, regular Pico will be also compatible but you'll need to download firmware file on your computer and put it onto the SD card manually. WIth Pico W it'll download the firmware for you.

boldandbrad, 2022-09-17:
I had the same issue and like others above I was able to resolve by shortening the wires!

All 5 of my wires are now 10-12 cm. My pico is mounted using the laser bear fan mount with dedicated pico slot - highly recommend - and I'm loading swiss from 256 GB SanDisk SD via SD2SP2.

I used 28 AWG wire because its what I had on hand. It seems people have had success with both 26 and 28 AWG.

Really excited and pleased with this mod so far!

boldandbrad, 2022-09-18:
Actually now that I've played with it a bit more I'm seeing inconsistent boot behavior. The gamecube doesn't consistently boot Swiss. More often than not, it boots to native bios.
It no longer just boots to black screen though - as it only did before I shortened the wires.

When it boots to swiss I am able to run homebrew software and games without issue. When it boots to native bios I can insert a disc and play without issue as well.

My expectation would be that if no disc is inserted the console would boot to swiss and if a disc is inserted, it would boot native bios and run the disc normally. But what I'm seeing is much more random.

Could it be that my wires are still too long? I don't have much slack to remove on them though. And I've quadruple checked my solder joints - everything looks correct - and I don't think swiss would boot and work at all if that wasn't the case.

The only other thing I can think of is maybe the SD2SP2 adapter is not having proper contacts with serial port 2 - it seems slightly loose.

webhdx, 2022-09-18:
There is a section in the wiki describing how to validate hardware installation. If you get iplboot output every time then it's your SD card setup.

Zonark, 2022-09-18:
> Actually now that I've played with it a bit more I'm seeing inconsistent boot behavior. The gamecube doesn't consistently boot Swiss. More often than not, it boots to native bios. It no longer just [...]

I had this same issue with mine found that the sd2sp2 adapter I was using wasn’t thick enough which cause intermittent disconnects. I bought the thing for really cheap my fix was actually buying a slightly more expensive one.

my other console the dol101 I made into a mini unit and SM just soldered a sd card to the board by connecting to the pin outs

boldandbrad, 2022-09-18:
Thanks for the replies!

I've ordered a (hopefully) better SD2SP2, a side loading one for access with gameboy player. I might try an SD Gecko also to see if there is a difference. I'm hoping that will resolve the intermittent issue.

But I do have another data point to share. I know this will sound crazy but in testing my install further I found the following:

After plugging the system into power and hdmi (via eon gchd mk 2), it always boots to swiss the first time. Sometimes, as described above, it boots to native bios after that. But if I remove the hdmi cable from the eon (or eon from the console, doesn't matter which) and replace it, it always boots to Swiss again at least once.

I have no idea why the hdmi connection would alter the boot state, but I was able to reproduce this strange behavior a dozen times without fail - very puzzling. I know the eon draws a little bit of power so my uneducated guess is that lingering charge isn't allowing pico to overload the boot process or something - but I admit I don't have a great understanding of what is at play here.

chippettone, 2022-09-24:
In my GC I installed the xenon chip. In order to mount the picoboot, do I have to remove the xenon chip? To avoid the problem of black screen can I use a shielding plate?

webhdx, 2022-09-24:
@chippettone [FAQ: Will it work if I have XenoGC installed?](https://github.com/webhdx/PicoBoot/wiki/FAQ#will-it-work-if-i-have-xenogc-installed)

werdferth, 2022-09-25:
I also have the same problem I shortened the wires to below 4in, but the light on the pico isnt on unless I directly cross the wire from gp5 to ground but it still doesn't give video output

webhdx, 2022-09-25:
@werdferth Post some pictures of your installation. As I said numerous times - black screen issues is usually caused by bad soldering or too long wires.

werdferth, 2022-09-27:
> @werdferth Post some pictures of your installation. As I said numerous times - black screen issues is usually caused by bad soldering or too long wires.

Right! Here hopefully they ste clear enough 

[image]
[image]
[image]

webhdx, 2022-09-27:
@werdferth It's pretty clear there are numerous shorts on Pico side where exposed end of the wire is touching adjacent pins. Soldering on GameCube side also looks suspicious to me but it's hard to tell from the photo. Redo Pico side and use a multimeter to ensure there are no shorts on the GameCube side.

renanbianchi, 2022-09-27:
GP7 seems to be shorting with GP8 and 3v3 is almost touching ADC_VREF, and GP5 seems to be touching GND. Cut your tips  and redo these soldering points.

ChipsLabs, 2022-10-26:
Got the same problem,wire were too long. It's working now with 13cm wire. Thanks everyone

srcorsario, 2022-10-28:
Hello.
Im having problems with a raspberry clone.
Tested with short and very short cables as seen in the photo
[image]

It boots to swiss on first try, however if i turn off and on the console it go to black screen.
I have to wait with the console off for about 10 seconds and Then I can turn it on and it goes to swiss perfectly.

If I use an original Pico it works without problems. Tested on different Japanese consoles.

Thanks in advance for your help

ChipsLabs, 2022-10-28:
> Hello.
> 
> Im having problems with a raspberry clone.
> 
> Tested with short and very short cables as seen in the photo
> 
> [image]
> 
> (Pin 6 and 7 bridged on the other side of the board)
> 
> 
> 
> It boots to swiss on first try, however if i turn off and on the console it go to black screen.
> 
> I have to wait with the console off for about 10 seconds and Then I can turn it on and it goes to swiss perfectly.
> 
> 
> 
> If I use an original Pico it works without problems. Tested on different Japanese consoles.
> 
> 
> 
> Thanks in advance for your help
> 
> 
> 
> 
> 
> 
> 
> 

How long are your wire ? They need to be aprox 13cm. Had the same problem and it resolve the issue

webhdx, 2022-10-28:
PicoBoot only supports the official Raspberry Pi boards. Never install any 3rd party board.

srcorsario, 2022-10-28:
@ChipsLabs
Wires tested from 13 to 5 cm and on differents japanese consoles.
On that picture the largest cable could be 5cm

@webhdx
It is understood that it is not supported. Thanks for answering

webhdx, 2022-11-01:
This issue became too lengthy. Black screen issue is well explained in the FAQ: https://github.com/webhdx/PicoBoot/wiki/Troubleshooting-tips#orange-light-is-on-but-it-doesnt-display-anything-on-the-screen

To sum it up:
- 99% of time it's caused by bad wiring or wires are simply too long
- in some cases replacing Pico might help
- never use any 3rd party board

If anyone experiences the same problems and can't resolve it with steps above please open new issue and post installation pictures.

closed 2022-11-01 by webhdx