webhdx/PicoBoot #40: Full IPL/BIOS Chip with Pico

ramon-code, 2022-09-18:
| Answer some general questions first                             	|  	|
|:-------------------------------------------------------------------	|:------:	|
| Are you my GitHub sponsor? | yes/no 	|
| Did you check older issues to see if it was requested before? | yes/no 	|

First, thank you very much for developing this beautiful project.

Now the request, I bought a Gamecube and it doesn't give video, even the laser moves, I changed the Bios chip and it worked perfectly. The problem is that there are no replacements. So maybe the PICO can fully function as a BIOS replacement.

webhdx, 2022-09-18:
It'll be very difficult to fully emulate RTC-DOL chip because it also includes RTC and SRAM. I want to eventually look into it but don't have high hopes for that.

masplund91, 2022-09-30:
A further question to this topic, so the IPL IC is still needed for the mod to work? I've read on a reddit post that we're just by-passing it, I though we utilized the chip with the pico? The pins we connect to on the IPL chip are the CS and serial out?

webhdx, 2022-10-01:
@masplund91 Yes you need a working DOL-RTC chip. PicoBoot only overrides IPL payload during initial bootup. DOL-RTC has more functionality to it - it handles RTC and SRAM too.

closed 2022-10-01 by webhdx