webhdx/PicoBoot #41: Eon GCHD Mk 2 and potential other digital converter incompatibilities

boldandbrad, 2022-09-18:
| Answer some general questions first                             	|  	|
|:-------------------------------------------------------------------	|:------:	|
| Which firmware version are you using?                             	| `v0.2` 	|
| Did you compile it yourself?                             	| no 	|
| Do you use alternative install method not mentioned in the guide? 	| yes (optional GND on Pico) 	|
| Are you using official Raspberry Pi Pico board?                   	| yes	|
| Did you pay someone to install it for you?                        	| no 	|

TLDR: EON GCHD Mk 2 causes strange picoboot behavior. Armor3 Nuview digital adapter (and maybe others), as well as analog outputs work fine.

After extensive testing of my picoboot install I have determined that certain GC digital adapters cause the GameCube to boot to native bios. Below I've documented my findings so that others may learn from them. Feel free to add this to troubleshooting docs or faqs.

My picoboot setup:

- DOL-001 USA GameCube w/ Serial Port 2
- Authentic Raspberry Pi Pico
- picoboot v0.2
- swiss v0.6r1362
- SD2SP2 adapter
- 256 GB SanDisk micro SD
- 10-12 cm 28 AWG wires
- Eon GCHD Mk 2 digital adapter

As I explained in [this thead](https://github.com/webhdx/PicoBoot/issues/37#issuecomment-1250356993), with the setup above the GC would boot to swiss with no issue the first time. In fact it would occasionally boot to swiss multiple times in a row. But eventually, without fail (usually after 2-4 successful swiss boots), the GC would boot to native bios. After booting to native bios once, it would only boot to native bios after that. Each time I was able to successfully solve the issue by unplugging and replugging in either the EON GCHD or the HDMI cable. Then it would start the described cycle over again.

As others suggested, I thought the issue was with my SD2SP2 adapter - I have one on order and will be able to test any differences with it later.

In the meantime I thought I'd try other possibilities, I learned in [this thread](https://github.com/webhdx/PicoBoot/issues/32) that it was possible for certain PSUs to cause similar issues. Luckily I have 4 GC PSUs, 3 DOL-002(USA) Made in China, and 1 DOL-002(USA) Made in Japan. But I experienced the exact same behavior with every single PSU.

Next I tested differences between output methods and this is where I found the problem. Here are the outputs I tried:

- EON GCHD Mk 2 - behavior explained above.
- Analog (GC Composite AV Cable) - no issues in 12 consecutive boots!
- [Armor3 NuView HDMI Adapter](https://www.gametronex.com/products/armor3-nuview-hd-adapter-hdmi-hdtv-upscaling-for-nintendo-gamecube?variant=39492614193237) - no issues in 12 consecutive boots!

To be honest I don't know the root cause of the Eons issues. I remember reading somewhere that someone had issues with digital out and picoboot, but I can't find that thread now.

But the Armor 3 digital adapter works just fine in my case.

It may be worth exploring other 3rd party adapters (Carby, etc) or mods (GC Video) to determine what is compatible with Picoboot and why.

Thanks!

C3x100, 2022-09-20:
I have a DOL-001, NTSC-J running with a US spec power supply. My TV connection is a EON Mk2 with otherwise identical setup to what you are running. My unit does not have SP2 so I am running off an SDGecko. I have not experienced this issue at all. just my two cents, it seems you have done a significant amount of testing but it seems more to me that your EON is the issue rather than the Pico install.

I have even been doing consecutive on and off boots sfter installing Cubeboot and messing with hex color code settings for my boot logo and have yet to have this happen.

boldandbrad, 2022-09-21:
With the only other difference being SP2 vs SD Gecko, it appears you're probably right. Either SP2 issue, which I'll test further when I get the new one/an SD Gecko. Or my EON is defective, considering yours works no issue.

And I agree, I don't think this is an install issue, but I wanted to report my findings in case there was a bug/incompatibility to document.

Thanks for the reply! Also thanks for introducing me to cubeboot! Definitely something I need to add to my setup :)

Extrems, 2022-09-21:
Note that there is two revisions of the EON GCHD Mk-II.

boldandbrad, 2022-09-21:
> Note that there is two revisions of the EON GCHD Mk-II.

I know there are firmware updates, and I know there was an original EON GCHD prior to Mk-II. But you mean there are two hardware revisions of Mk-II itself?

I just tried googling and I'm not finding any evidence of this. What is the difference and how can I verify?

Extrems, 2022-09-21:
Rev 2 changed the FPGA's package from a QFP to a BGA, added a JTAG header alongside the existing SPI header, and flipped the video DAC and FPGA's places.

rokoji18, 2022-11-01:
I am having the same issue with the EON GCHD MkII when using SD2SP2. I pulled out a CRT and tested with the composite cables and it worked about 10 times in a row with no issues. With the EON GCHD MkII I was getting maybe 1 in 3 startups loading correctly. 
Prior to this I have also tried other power adapters, a few styles of SD2SP2 adapter, an SD Gecko, and unsoldering and re installing with a new pi pico with no real results.
[image]
I appear to have the rev 2 version with the BGA FPGA.

C3x100, 2022-11-01:
> I am having the same issue with the EON GCHD MkII when using SD2SP2. I pulled out a CRT and tested with the composite cables and it worked about 10 times in a row with no issues. With the EON GCHD [...]

I am running an identical setup on mine but havent had a single boot issue. I ran it with the standard boot.dol for a while and now use Flippyboot / Cubeboot for the added start screen.
My hardware is a Pico + EON Mk2 on a NTSC-J DOL-001 (without SP2) + an SDGecko.

Possibly a bad EON?

webhdx, 2022-11-01:
In my opinion GameCubes are starting to show their age and they are now prone to all kind of different power issues caused by failing capacitors. I had very similar issue with another GCVideo based cable, it wouldn't boot every time with video. To fix the issue I replaced power regulator daughter board. Most likely capacitors were failing on that one. If you have another GameCube I'd recommend swapping the power board.

rokoji18, 2022-11-02:
Swapped the DC board on the left for the one on the right from another GC, no failures with the new board. I do plan on installing picoboot on the one that I took that DC board from so I'll re-cap it at some point.
[image]

closed 2022-12-03 by webhdx