webhdx/PicoBoot #42: No 3V from GameCube

hosser, 2022-09-19:
| Answer some general questions first                             	|  	|
|:-------------------------------------------------------------------	|:------:	|
| Which firmware version are you using?                             	| `v0.2` 	|
| Did you compile it yourself?                             	| no 	|
| Do you use alternative install method not mentioned in the guide? 	| no 	|
| Are you using official Raspberry Pi Pico board?                   	| yes 	|
| Did you pay someone to install it for you?                        	| no 	|

Hi. I'm having trouble. I am not getting power to the Pico from the GameCube - the LED on the Pico does not light up when I power on the GameCube. If I desolder the wire from the 3V3_EN pin on the Pico and connect a USB power supply at the same time as powering on the GameCube, the Pico LED lights up, I see PicoBoot boot messages and Swiss loads successfully. So I can rule out all other issues - I have a power supply problem. 

I have soldered the 3V3_EN pin on the Pico to the correct pin on the (DOL-001) motherboard according to the installation diagram. The joint is good and I have tested continuity with a multimeter. I shortened the wire length so that it is now under 4 inches as I read this can cause issues. 

If I connect the USB power supply to the Pico when the 3V3_EN wire is soldered, the LED does not light up. Is this normal? Does having a wire soldered to 3V3_EN pin disable power input from USB?

Is there any difference to the pinout for PAL GameCubes? I have a day 1/launch model DOL-CPU(P)-01 motherboard. Does 3V come from elsewhere on this board? 

Any other ideas please? I can post photos of anything required.

webhdx, 2022-09-19:
You have to solder to 3V3 point, not 3V3_EN.

hosser, 2022-09-19:
> You have to solder to 3V3 point, not 3V3_EN.

God what an absolute chimp I am. I didn't even notice the other 3v3 pin despite having pinout.xyz in front of me (not to mention the labels on the damn thing I am soldering). I'll try this tomorrow - thanks. I'm sure that's all I've done wrong. Thank you.

closed 2022-09-19 by hosser