webhdx/PicoBoot #44: cannot flash picoboot.uf2 v.1 or v.2

Eyeball38, 2022-10-02:
hello. i am having a problem flashing the picoboot file to my pico W.

i have tried on a windows 10, ubuntu destop and also a pi4 all with the same result.

each computer recognises the pico when boot select is presed and wen i copy either of the two version files it appars to be copying, disconnects but no green LED lights on the board.

on reconnection to either computer the pico acts as if nothing has been copied.

i have used the nuke_flash.uf2 file and retried to same effect.

i was leaning to faulty pico but have succesfully flashed the circuitpython .uf2 file from adafruit with no issues, which rule out a board fault.

could you please help me as the web does not seem to have an aplicable anwsers.

kind regards

bart1ebee, 2022-10-02:
Known issues with the Pico W https://github.com/webhdx/PicoBoot/issues/13#issuecomment-1178602252

webhdx, 2022-10-03:
Answer to your question is in the FAQ: https://github.com/webhdx/PicoBoot/wiki/FAQ#can-i-use-new-pico-w-board

closed 2022-10-03 by webhdx