webhdx/PicoBoot #46: Lost C169 during installation process

DelBoss, 2022-10-14:
| Answer some general questions first                             	|  	|
|:-------------------------------------------------------------------	|:------:	|
| Which firmware version are you using?                             	| `v0.2` 	|
| Did you compile it yourself?                             	| no 	|
| Do you use alternative install method not mentioned in the guide? 	| no 	|
| Are you using official Raspberry Pi Pico board?                   	| yes 	|
| Did you pay someone to install it for you?                        	| no 	|

Hey guys, 

I installed the Pico board, got a blackscreen and desoldered the Pico what made the GC start again. I then fixed bridged pins and resoldered the Pico, got a blackscreen again and then noticed that I somehow lost the C169 capacitor.

I got very calm hands and have no clue how I fucked that up. Im still very confident to fix this myself but dunno what kind of capacitor I will need. Does anybody know the type and where to buy these caps? I got a broken GBA. Is there one of these caps on a GBA?

Sorry for my bad English, I'm German :D 

Thanks in advance
[image]

KainCo, 2022-10-14:
No the 169 is free (no cap).
Nein der 169 ist frei, war bei mir auch so.

DelBoss, 2022-10-14:
> No the 169 is free (no cap). Nein der 169 ist frei, war bei mir auch so.

Oh really, thats not good though, then I fucked sth else up. :/ Why did I get the GC to start after holding bootsel or desoldering the Pico and now it doesnt work anymore?

KainCo, 2022-10-14:
you broke the outer pin at U10, if I see correctly. Take a look and repost if it is broken or not.

webhdx, 2022-10-14:
Looks like you destroyed U10 which is a shame because you can't buy a replacement. This motherboard is a toast 😢

DelBoss, 2022-10-14:
Shit. Would it be possible to buy a broken Gamecube and use a U10 chip of this?

webhdx, 2022-10-14:
Sure. Buy console with a dead disc drive and replace the whole motherboard.

DelBoss, 2022-10-14:
Do you mean the U10 chip by package? So scrape a bit where the leg joins the chip? What do I have to do then?

This might be a stupid question but could I try to get the splder to join the U10 chip? I could imagine this will cause shortings, wouldnt it?

KainCo, 2022-10-14:
Maybe you can fix it by resoldering the Lag correctly and press it gently to the chip. Or reglue the lag. Electrical paint also can be an option

DelBoss, 2022-10-14:
I mean whatever I can do is worth a try, isnt it? Have a working or a broken Gamecube :D

KainCo, 2022-10-14:
Learn from mistakes. greater the joy when you get it repaired :D

DelBoss, 2022-10-14:
I know its very hard to see but isnt there some exposed metal in the chip? What would happen if I try to get solder straight to that point will or wont there be short circuits? Do you zhink I should try this or sth else?

KainCo, 2022-10-14:
Yes there is an expose metal. I would try to over paint it with electrical paint. That's the way to go for a beginner. But you can also try to get a good load of soder to the exposed point but it is very difficult and you risk a short or a second damage.

DelBoss, 2022-10-14:
Okay so Ill order some electric paint and try. But what will I have to do? Desolder the leg or the chip or just keep it how it is and use the paint to get a connection?

KainCo, 2022-10-14:
the color is a kind of lacquer, mostly like nail polish.  remove the leg and then over paint the exposed metal and draw a line down to the connection where the leg normally sits.  The chip itself, (the black one,) is not conductive

DelBoss, 2022-10-14:
Ahh, so just remove the leg and solder and use the color instead. Thank you very much, ill give it a try. 

If this doesnt work, ill try with solder and if this doesnt work too, just buy a GC with a broken disc drive. I think ill desolder the 3.3V wire before to check if it was succesful cause I dunno if the mod worked.

Thanks for the tips and also thanks for the hawkeyes that spotted the broken leg, I would have never seen it. Also thank you webhdx for helping and providing such an awesome mod. :D

renanbianchi, 2022-10-14:
You have way too much solder on those pads. Use a desoldering braid or a desoldering pump tool, and flux to make the solder behave while doing it and remove that excess solder.

On the leg issue, i would advise using a dremel to remove a little of the plastic cap to unveil more on the leg region and restoring the pad.

KainCo, 2022-10-14:
You're welcome.  but be careful, the paint could be very runny depending on the manufacturer.  therefore test the behavior on a piece of paper and use a toothpick to apply.  feel your way slowly.  If you have any questions, you can also contact me on reddit.  have the same username there.

webhdx, 2022-10-14:
As much as I'd love you to repair it yourself you need to realize it'll be incredibly more difficult than installing PicoBoot itself. You should get a replacement motherboard and find someone who will install PicoBoot for you. Unfortunately you can't attempt any soldering project having no soldering experience whatsoever. If you consider to ever attempt it again - do yourself a favor and practice on some cheap SMD soldering kit or even scrap electronics.

DelBoss, 2022-10-14:
I know that but I dont get the point. If I dont try it, the motherboard is surely dead. If I try it, there is at least a small chance of success, even if its unlikely that I will succeed. Either saving 40€ or not.

KainCo, 2022-10-14:
you should try it.  This is the best example to learn from.  That's how I learned my skills.  You have nothing to lose and you can practice.  Yes there is too much solder on the legs of the chip but that doesn't affect the function.  don't take it away is my opinion.  The solder spots also look pretty cold you should solder at a higher temperature.  The solder should shine.

DelBoss, 2022-10-14:
I put my soldering iron to 360°C. What points do you mean? Would it be bad to just leave it how it is? 

I still dont know why webhdx wants to demotivate me from trying to rescue a motherboard and wants me to buy a new one but then says I should practice on sth else. I think if you have nothing to lose, thats the best to practice with.

renanbianchi, 2022-10-14:
Webhdx has seen too much botched installs, seeing that everyday can mess you up. Jokes aside, don't worry, you'll eventually be able to restore it.

KainCo, 2022-10-14:
I don't think webhexhd wants to demotivates you.  he just wants to save you some hassle and work.  In my opinion, 360° is too high.  rather go to 300. SMD parts in particular are very sensitive.  electric paint in German: Elektrisch leitfähige Farbe oder Graphitlack. don't worry leave it as it is for now and fix the chip

renanbianchi, 2022-10-14:
Rule of thumb i use: Never use force, never touch any point with your soldering iron for more than 6 seconds.

DelBoss, 2022-10-14:
Thanks for the tip, I often heard that the best temp for sth like this is 300°C. 

But please be honest: Do you think I can fix it? Is there a way to correct if the paint goes where it shouldnt go?

KainCo, 2022-10-14:
Look... it's fixable, but I don't know if you can do it with your current skills and tools.  But i can give you the advice to try it as you have nothing to lose.  watch some YT videos on how to fix this and give it a try.  if the paint runs you can also wipe it away as long as it has not dried (takes about 10 minutes) You are also welcome to register in this German forum: https://circuit-board.de there are many modders who could give you good tips for your problem

DelBoss, 2022-10-19:
So update: I tried to fix the broken pi  with Bare Conductive but unfortunately, it didnt work. I dunno if there is any way to fix it anymore.

KainCo, 2022-10-19:
Broken PI?  wasn't the IC on the mainboard of the GC broken?  Can you post a picture

webhdx, 2022-10-20:
As I said it'll be almost impossible to fix the IC at this point. Better get another board and save that one for the time where we have U10 replacement. If you torture the board too much and destroy traces or pads it'll be impossible to fix.

closed 2022-11-01 by webhdx