webhdx/PicoBoot #47: Games constantly freeze on me along with a beeping sound

Lote99, 2022-10-19:
Hello, first of all thank you for making this project possible. I have had many problems with the games which freeze along with an annoying beep I have changed the solder cables many times, the sd2sp2 I have soldered and it works very well, specifically this error gives me on the 001 gamecube, I have heard that this error is due to overheating or problems with the thermal pads, I have installed picoboot on the 101 models and I have not had any problems.

webhdx, 2022-10-20:
Does it also happen on disc games? If so overheating may be an issue here. If not then I'd recommend trying different adapter and SD card. Make sure it's FAT32 formatted.

Lote99, 2022-10-20:
> 

This gamecube does not have a disc reader so I cannot test it, I have used that same sd in another gamecube 101 and it works very well, I do not know where the problem lies

webhdx, 2022-10-20:
You should try out a different SD adapter. You couldn't have used SD2SP2 on DOL-101 because it doesn't have SP2 port so the only differentiating factor here is the SD adapter.