webhdx/PicoBoot #48: Component desoldering unintentionally (Two left hands)

NicolasMKZ, 2022-10-22:
| Answer some general questions first                             	|  	|
|:-------------------------------------------------------------------	|:------:	|
| Which firmware version are you using?                             	| `v0.1` 	|  ???  Dol-001
| Did you compile it yourself?                             	| yes/no 	|  No
| Do you use alternative install method not mentioned in the guide? 	| yes/no 	|   No
| Are you using official Raspberry Pi Pico board?                   	| yes/no 	|  Yes
| Did you pay someone to install it for you?                        	| yes/no 	|   No

Hello I soldered the chip yesterday and without seeing it by turning over the motherboard I unsoldered 1 resistor behind (how I did I don't know) and I tried to resolder it but it was coated with tin already coated on the pane so I would like to know the value of this resistor to buy it back and thus be able to repair my console?

The resistor on the back (Side-B) is resistor R3 if someone can measure their values ​​and give them to me I would appreciate it and if he can give me directly the reference of a resistor that could replace it would be even better ^^

Sorry to bother you because I'm not very good with my hands sometimes T-T

On the (Side-A) I also think I damaged the capacitor "C170" and the unknown component noted "X2" but I don't know

I have a black screen when I start with the PicoBoot no sound and even when unsoldering the PicoBoot the console starts perfectly but no sound or images just a black screen!

Thanks you 🥺

webhdx, 2022-10-22:
You need to post pictures, without them it'll be difficult to evaluate the damage you've done.

spitford, 2022-10-24:
You could probably use some solder braid to clean up the old resistor/any extra solder.
i.e. https://www.amazon.com/Lesnow-No-Clean-Desoldering-Removal-Dispenser/dp/B094GZ6CPZ/ref=sr_1_5

webhdx, 2022-11-15:
No activity during last 22 days. Closing.

closed 2022-11-15 by webhdx