webhdx/PicoBoot #49: Killed IPL Chip

EliteNinjaX, 2022-10-24:
| Answer some general questions first                             	|  	|
|:-------------------------------------------------------------------	|:------:	|
| Which firmware version are you using?                             	| `v0.1` 	|
| Did you compile it yourself?                             	| no 	|
| Do you use alternative install method not mentioned in the guide? 	| no 	|
| Are you using official Raspberry Pi Pico board?                   	| yes|
| Did you pay someone to install it for you?                        	| no 	|

I killed some of the pins on the IPL, and I don't know if I can save it or not. I was kind of thinking that maybe there is another pinout perhaps, or maybe get some advice about what to do from someone smarter then me.
[image]

webhdx, 2022-10-24:
There is no way to fix it and no replacements are available because it's a proprietary chip. This board is a toast. Please mind the warning next time - the documentation clearly states to practice soldering before attempting to mod a GameCube.

closed 2022-10-24 by webhdx

EliteNinjaX, 2022-10-26:
Yep that's my fault. I did practice some, but I was using a cheep Iron, and should have knew better.  I do have an extra board and will try again after I get a better Iron and practice even more. I won't give up with this project or even fixing this board just yet, thank you for your time and this project.