webhdx/PicoBoot #51: GC not booting

Marthswordmaster, 2022-11-03:
| Answer some general questions first                             	|  	|
|:-------------------------------------------------------------------	|:------:	|
| Which firmware version are you using?                             	| `v0.1` 	|v0.1
| Did you compile it yourself?                             	| yes/no 	|no
| Do you use alternative install method not mentioned in the guide? 	| yes/no 	|no
| Are you using official Raspberry Pi Pico board?                   	| yes/no 	|yes
| Did you pay someone to install it for you?                        	| yes/no 	|no

When turning on
[image]
[image]
[image]
 the Gamecube it wont start but if I desolder GP4 form the ipl it starts and asks for calendar setup.

bart1ebee, 2022-11-03:
It is hard to tell how long the wires are it is the most common problem they're too long.

Marthswordmaster, 2022-11-03:
How long should they be?

> It is hard to tell how long the wires are it is the most common problem they're too long.

How long should they be?

Marthswordmaster, 2022-11-03:
I measured them and they are like 28cm long

renanbianchi, 2022-11-03:
[image]

Check if there is a short in this region.

renanbianchi, 2022-11-03:
> I measured them and they are like 28cm long

There isn`t a specific number for wire lenght that i know of. Try to reduce the wiring to about 20cm or less.

bart1ebee, 2022-11-03:
Here in the wiki 10cm or less https://github.com/webhdx/PicoBoot/wiki/Troubleshooting-tips#orange-light-is-on-but-it-doesnt-display-anything-on-the-screen

Marthswordmaster, 2022-11-03:
> Here in the wiki 10cm or less https://github.com/webhdx/PicoBoot/wiki/Troubleshooting-tips#orange-light-is-on-but-it-doesnt-display-anything-on-the-screen

Thank you very much, I will change the length of the cables and see if it fixes it

Marthswordmaster, 2022-11-03:
Hi I already changed the lenght and is now working, but I got no audio

webhdx, 2022-11-03:
Audio has nothing to do with PicoBoot. If you are using GCVideo based HDMI adapter make sure "enhanced DVI" is enabled.

Marthswordmaster, 2022-11-03:
> Audio has nothing to do with PicoBoot. If you are using GCVideo based HDMI adapter make sure "enhanced DVI" is enabled.

Im using the original av cable, it was working some minutes ago

webhdx, 2022-11-03:
PicoBoot doesn't interfere with audio in any way. If you are absolutely sure it worked on your setup prior to PicoBoot installation, it's plausible you damaged something during the install.

closed 2022-12-03 by webhdx