webhdx/PicoBoot #53: Swiss no Booting and not sure at this point what is going on.

Akurie, 2022-11-11:
| Answer some general questions first                             	|  	|
|:-------------------------------------------------------------------	|:------:	|
| Which firmware version are you using?                             	| `v0.2` 	|
| Did you compile it yourself?                             	| yes 	|
| Do you use alternative install method not mentioned in the guide? 	| no 	|
| Are you using official Raspberry Pi Pico board?                   	| yes 	|
| Did you pay someone to install it for you?                        	| no 	|

I solder all the out puts that are required from the Pi to the GC locations.  I formated the SD card to FAT32 and followed the instructions in naming the .dol file to ipl.dol and the native ipl keeps booting up.
[image]
[image]
[image]

webhdx, 2022-11-12:
1. https://github.com/webhdx/PicoBoot/wiki/Troubleshooting-tips#my-console-boots-to-vanilla-gc-menu
2. It may also be caused by your soldering. Some of these points looks like they could've been touched up. Especially the ground wire which looks to be hardly soldered. You are using quite thick wires, even though this is generally okay I'd recommend using slightly thinner ones that would be easier to solder to such small points.

Akurie, 2022-11-12:
I'll Cleaning up my solder and get some clean wire.  the size I am using is 24 gauge might look big because the picture and some of the jacketing is getting hot will using lead free solder.

webhdx, 2022-11-15:
Lead free solder requires much more heat and in hands of beginners is usually a recipe for a disaster. Despite all the health concerns I'd recommend using leaded solder. Also do not use AWG24 wires, they are too thick! AWG26 at maximum, AWG28 might be even easier to solder. Be aware of single loose strands which can short to adjacent pins.

closed 2022-12-03 by webhdx