webhdx/PicoBoot #54: Pico seemingly ignored by console

aarmastah, 2022-11-12:
| Answer some general questions first                             	|  	|
|:-------------------------------------------------------------------	|:------:	|
| Which firmware version are you using?                             	| `v0.2` 	|
| Did you compile it yourself?                             	| no 	|
| Do you use alternative install method not mentioned in the guide? 	| no 	|
| Are you using official Raspberry Pi Pico board?                   	| no 	|
| Did you pay someone to install it for you?                        	| no 	|

I have an unofficial board that I have confirmed works in another GameCube console, trying to use the same model in this one is leading me nowhere. I tested continuity between my solder points and they all look good, yet the console only boots normally and will not display the debug output nor boot from SD. Anything I'm missing potentially?
[image]
[image]

webhdx, 2022-11-12:
1. Your wires are waaay too long.
2. As stated in FAQ you won't receive any support if using unofficial board.

aarmastah, 2022-11-12:
> 1. Your wires are waaay too long.
> 2. As stated in FAQ you won't receive any support if using unofficial board.

1. Are they? I'll try shortening them to see if it helps.
2. That's totally fair lol. Just figured I would give it a shot anyway since this type of clone board worked in my other GC.

webhdx, 2022-11-12:
I'm not even sure if all pins on that board match with the official one. With tens or hundreds of different RP2040 boards I can't support and test them all.

Check basic things first - is your Pico getting power?

aarmastah, 2022-11-12:
> I'm not even sure if all pins on that board match with the official one. With tens or hundreds of different RP2040 boards I can't support and test them all.
> 
> Check basic things first - is your Pico getting power? 

The pico definitely gets power. When I turn the console on, it lights up red for PWR, same as when plugged in via USB.

I just shortened my wires and sadly am getting the same results.

[image]

webhdx, 2022-11-13:
I wouldn't use that board. My firmware is bootstrapped for particular flash chip. If this board uses a different one it'll have problems. I'm not even sure if it's pin compatible with the official Pico board. Just get the recommended one and you'll be good.

aarmastah, 2022-12-02:
> I wouldn't use that board. My firmware is bootstrapped for particular flash chip. If this board uses a different one it'll have problems. I'm not even sure if it's pin compatible with the official Pico board. Just get the recommended one and you'll be good.

I purchased some official boards and it worked perfectly first try. Not sure why my first bootleg worked fine and the others didn't, but I'm just glad it ended up working out. Thanks!

closed 2022-12-02 by aarmastah