webhdx/PicoBoot #57: Add support for Pico W and handle discrepancies between the boards

webhdx, 2022-11-15:
Pico W debuted the day PicoBoot was released. It gave me no time to prepare codebase for Pico W. This is why I never advertised PicoBoot as fully compatible with Pico W. As for now it's working correctly without the support for status LED which is not a big deal.

My plan for Pico W:
* release single *.uf2 firmware file compatible with both boards
* distinguish between the boards based on the presence of WiFi module (I've already done it in my tests)
* keep core functionality the same
* expose WiFi module over EXI bus so homebrew can communicate with the internet through PicoBoot (blocked by: TBA)
* make Pico W "the go to" platform for PicoBoot and advertise WiFi capabilities as the main selling point and a pillar for modernized homebrew ecosystem on GameCube

sli, 2023-01-30:
I would really like to suggest making the Tiny2040 a first class citizen in this project due to its size relative to the Pico and Pico W.

webhdx, 2023-01-30:
Unfortunately as I stated many time I'm not planning to extend support to other RP2040 boards. This is simply because I don't have enough time to test different boards to make sure new firmware is going to work for everybody. I heavily overclock Pico in order to pull off IPL injection and for this you need very repeatable hardware. With official boards I can be assured every batch uses components that follow the specification. WIth 3rd party boards it's not always the case.
PicoBoot adopt Pico W as the main platform and heavily utilize CYW43439 WiFi chip. Tiny2040 and other Pico boards don't have that specific chip.

webhdx, 2024-08-05:
I've learned about WaveShare RP2040-Tiny board:

[image]

I've been against supporting unofficial boards but this one is particularly interesting because it externalizes USB port (it's also USB-C!) and buttons via flat flex cable. Many people have been asking about putting USB extender for easier updates. This perfectly addresses this issue. The biggest issue I see is Nanopixel LED which requires PIO. We can afford it at the moment but we might run out of PIO resources when more advanced features are implemented in the future.