webhdx/PicoBoot #59: Make PicoBoot fully addressable and controllable via homebrew

webhdx, 2022-11-15:
When I started doing my research on IPL injection technique, I mostly based my research on [Shuriken Attack](https://retro-system.com/shuriken_attack.htm) uses quite basic method. In short - it listens on boot sequence and counts bits transferred to injecting its payload at the exact moment. This should be revamped to fully support reading all the commands issued to IPL chip. This will not only improve IGR (PicoBoot doesn't work on subsequent resets) but also let homebrew communicate with PicoBoot. This means we will be able to read LFS filesystem (#56), update the firmware using DOL file (as opposed to the current method requiring a USB cable) and use WiFi module on Pico W (#57).

I've made some initial tests with Pico talking to homebrew via EXI bus and communication is not really a problem here. The biggest problem to overcome is needing to transmit data on 4 pins at the same time. This is required to electrically overpower IPL chip as it's working at the same time. 2 pins are enough to do that but I couldn't make it work when data was outputted on 2 pins, it required 4... 

We ultimately want to keep payloads in raw form (currently they have to be bit duplicated) and maybe use 2 different PIO SM's to output the same data on 2 different pins. 

This feature is the top priority and it'll be crucial for all other changes.

jmsunseri, 2024-04-30:
How can someone help with this?

inspectah88, 2025-01-13:
If this feature gets implemented and firmware updates via DOL files would be possible, would that also remove the need to disconnect 3.3V wire for the time of update?
(I know lots of ifs and woulds)