webhdx/PicoBoot #60: Make companion app to configure PicoBoot

webhdx, 2022-11-15:
When #59 is implemented it'll open up the door to talk to PicoBoot from GameCube side. Some kind of companion app would be needed to check for firmware updates, configuring network settings etc. 

The app itself will be a huge app but we should identify milestones and features. I don't think the app is first necessity so we can put it in v2.0 milestone.

## General list of features
* show PicoBoot version and board
* detect unsupported board variants and warn users (AliExpress kits)
* check for online updates
* download and perform firmware updates
* check for new Swiss version/download it? (maybe this should be implemented into Swiss)
* allow firmware updates from uf2 file stored on SD card (aka offline update)
* provide configuration for cubeboot?
* ?

## First app milestone
* show PicoBoot version and board used
* online and offline updates

KiritoDv, 2022-11-15:
I could help on the companion app development, a cool idea could be to use flutter

webhdx, 2022-11-15:
Sorry maybe I wasn't precise - the app is meant to run on GameCube. We are talking C and libogc here.

KiritoDv, 2022-11-15:
Oh sorry, but it would be cool a feature like that anyway