webhdx/PicoBoot #62: Finalize pinout and release flex cable

webhdx, 2022-11-15:
# Pinout
It's known that PicoBoot is still experimental and I've been vocal about upcoming wiring change. This needs to happen in v1.0, should be finalized and never changed again. 

Current wiring doesn't utilize MOSI signal at all but this will be critical for other features to work. I briefly talked with Extrems on providing WiFi access and he requested to use interrupt signal from SP2 to optimize communication.

# Flex cable
This has to happen. There are some 3rd party solutions available but IMO they are harmful as no one pays attention to how PicoBoot is wired and why you need to disconnect 3.3V line during updates. This was done to simplify the installation for 99% of users but this should be definitely addressed by flex cable as you don't have any way to disconnect power from the console. 

There are differences between DOL-001 and DOL-101 models and I don't think single flex cable can support both models. So we are talking about 2 different revisions of the cable.

PCB quick solder board is not going to work as some of the other modchips tried and failed IMO. Custom flex cable should be used instead. It's going to be more expensive but I believe people will prefer to pay slightly more to have professional and robust setup.

mdr92, 2022-11-22:
This is the big one for me, presuming the new wiring will support the existing code, I've held off installing mine until the new wiring is finalised. It'd be good to get it installed and working even without the new features (which look great).

webhdx, 2022-11-24:
v0.1 and v0.2 firmwares will be only compatible with current wiring method. For any upcoming firmware you'll need to use new wiring method. Unfortunately it won't be possible to provide support for both wiring methods. This is a change that is needed now as all new features are basically not possible with current setup :(

dtomcat, 2023-01-09:
Any update on this?  Thank you

webhdx, 2023-01-09:
@dtomcat No. I'm taking a longer break from PicoBoot currently. I have other projects to finish first.

brian4120, 2023-04-06:
Thinking offhand, would placing a small diode between the GC and 3.3v input help prevent backfeeding the GC while the pico is powered via USB? I had a similar issue with a project I was working on but I was using a DMG2305UX P-Channel Mosfet to keep from backfeeding the USB power per the Pico's Datasheet.

webhdx, 2023-04-06:
@brian4120 This is 100% correct and in the future I'm going to make a diagram how to correctly include a P-Channel MOSFET to resolve that issue. I didn't want to complicate installation process initially because finding correct diode and wiring it with correct polarity would be way too difficult for many people. Heck, a lot of ppl are struggling soldering 5 wires and they break IPL chip in the process.

mdr92, 2023-06-08:
Any updates? Appreciate you may not have time to work on this at the moment, but it would at least be nice to get finalised wiring so I can do an install and know I won't need to change it, even if additional wires are needed for other functions (networking) down the line? Thanks

webhdx, 2023-06-08:
Nothing is confirmed at the moment, sorry. I'm planning to revisit PicoBoot when I finish working on other projects.

nex86, 2025-04-27:
I'm working with the flex cable, but can't get any power on the pico.
does anyone have the pinout for the IPL chip so I can see where the 3.3V actually are?

webhdx, 2025-04-27:
Are you using aliexpress flex cable? It has 2 contact points on the back. You have to solder bridge them.