webhdx/PicoBoot #63: Turn PicoBoot into full DOL-RTC replacement

webhdx, 2022-11-15:
PicoBoot's popularity contributed to a huge pile of broken consoles as people who's never soldered anything are now trying to install it without having any experience in modding/soldering. Usually they manage to break IPL chip which unfortunately is not replaceable. By far no one came with replacement chip that could fix all dead GameCubes.

I've started researching what could be done to provide DOL-RTC replacement and it may be possible with Pico.

The original IC has multiple functionalities:
* ROM
* RTC
* SRAM
* AD16 (some kind of boot process logging device)

This could be potentially implemented on Pico in the future. Prior to that we definitely need to have: #62 and #59.

Should PicoBoot even cover this? I feel like it shouldn't be mixed with PicoBoot, maybe separate project would be better and could utilize custom PCB?

### ROM
This is the easiest part as this is what PicoBoot is basically doing at the moment

### RTC
This one is tricky. RP2040 has builtin RTC so it doesn't require external modules but you still need to keep Pico powered by a backup battery. The question is how long can you power Pico from cell battery? I haven't checked any deep sleep modes yet but if I had to guess it'll deplete the battery much faster than original IC. My another idea was to eliminate the battery completely and rely on setting up the clock with NTP via network. This seemingly won't work as you'd need a few seconds before NTP could be accessed (setting up network, DHCP, DNS all add up).

### SRAM
Easy to emulate with SRAM available on Pico. It could be also backed up in flash.

### AD16
I haven't looked into it in details but it seems to be fairly simple to emulate.

webhdx, 2022-11-15:
Thanks, updated the description.

Grayda, 2022-11-22:
Not sure if I'm missing something here, but could you just set a "default" time and date (e.g. 1/1/2000) to avoid the RTC boot penalty, then just update the clock once you've got the time from an NTP server? If there's only a few seconds delay between boot and getting the time, I doubt most people would notice.

the0remora, 2023-02-22:
So this would allow the clock to go past 2030?

Extrems, 2023-02-22:
The clock already goes up to 2136.

brucenovaa, 2024-09-08:
Hi!!! Thank you for the great work put in this project. 
I saw that there's a roadmap for the picoboot, but this one is way below on priority, and would like to know if there's anything that I could do to help here. I bought a bunch of GameCubes with busted ipl-rtc chips, and helping finalize this roadmap would benefit me a hundred more. (Sorry for my bad English)