webhdx/PicoBoot #65: Run webserver with configuration/status page

webhdx, 2022-11-15:
90% of the time WiFi module on Pico W will idle. Since we have second core available we can run small webserver showing status page with diagnostic information. It could be also possible to provide another configuration interface. This is not in any way important but may be still a cool feature to give PicoBoot more edge over other modchips.

SpiLiam, 2022-11-15:
approved

Mello5617, 2023-03-13:
This'll be cool

pjc09h, 2023-06-03:
I'd throw some weight behind this, especially if we intend to run an authenticated Samba *client* rather than the proposed server and sync them to local M.2 SSD drive files, so as not to load games off a shitty Pico W wireless connection.

The idea in my mind is a configuration page that manages a Samba share to sync on boot. So you can point the modchip at your GazelleGames seed folder on the LAN and automagically have your recent game downloads locally available in due time.

I guess it should also spit out whatever system stats are available, but you'd need to guide me through the hardware interface to get stuff like temperature, fan speed, etc. Senior PHP developer here, not an electronics engineer.

One last thought: TOTP or at least basic HTTP auth is essential. My main gripe with similar things like the MemCard Pro, is that all your data is exposed on the LAN unauthenticated.

webhdx, 2023-06-03:
Samba client makes a lot of sense. Thanks for the idea!

pjc09h, 2023-06-03:
I return with probably bad news.

[MicroPython received wisdom](https://forum.micropython.org/viewtopic.php?f=2&t=3894) says that a Samba client in that language is a non-starter, and to use FTP/HTTP instead. It may be possible in C with [existing libraries](https://github.com/samba-team/samba/blob/master/source4/client/client.c), but given the [slow EXI bus speed](https://www.reddit.com/r/Gamecube/comments/q60uvf/comment/hg92hy9/), I dunno how useful it would be or what happens if a Samba transfer is stopped and resumed.

How feasible would it be to integrate a USB port into the M.2 Loader as in [this kind of board](https://www.amazon.com/dp/B01NH2W8NL?th=1), so as to mount the drive on a laptop out the side of the GameCube? I saw in the prototype video that the Micro-USB port would need to be where the M.2 port currently lives, probably limiting the SSD form factor and causing a lot of headaches...

Anyway, maybe a good use for this web page would be to back up and restore Swiss configurations, upload new PicoBoot firmware, and to view and save diagnostic logs?

webhdx, 2023-06-03:
USB is feasible but RP2040 only supports USB 1.1, it's slow... too slow for transferring ISO images.

Extrems, 2023-06-03:
Why do they need to be transferred? They can already be loaded directly from the network.

pjc09h, 2023-06-03:
The thought was more like an integrated USB 2.0+ adapter that bypasses the RP2040, for the sole purpose of mounting the drive on a laptop without external hardware. Extrems, buying a BBA doesn't thrill me.

That discussion is outside the scope of the issue. Back on topic, I'd be interested to make a small web page similar to IPMI. Something with authentication over self-signed TLS if possible, that accepts firmware uploads and shows an SHA hash of the upload before flashing it and rebooting, can transfer small files like Swiss configs, and display basic stats.

There's a couple extra things like making an access point to initially connect to, and associating the Pico W with an existing network and persisting the info. OTA updates in particular seem sketchy beyond the point of "verify the received hash and don't shut down the device while flashing."