webhdx/PicoBoot #69: Picoboot, cannot boot into Swiss anymore

rakebi, 2022-11-30:
Hi!
I modded my 001 Gamecube with Picoboot, tried to add the Cubeboot but it didn't work (swiss would not boot and got an error). Got rid of Cubeboot and put back Swiss dol and everything worked fine - excluding the awful quality of the video. 
Now, I got a digital hdmi adapter (Bitfunx), it works and the video quality is pretty good. The small issue is that Swiss won't boot, instead the normal Gamecube dol boots. I switched back to the AV cable and Swiss is still not booting.
I'm using a sd2sp2 and also tried with a gc2sd. The pico board has the green light.
I want to add that before plugging the hdmi adapter everything worked fine and I got 15h+ of gaming.
Thank you!

webhdx, 2022-11-30:
You'll need to do some troubleshooting first.

1. Make sure you are using the official firmware from this repository. You can download `picoboot.uf2` from here: https://github.com/webhdx/PicoBoot/releases/latest I don't support CubeBoot and I recommend using DOL version instead of their uf2 firmware.
2. Once you have v0.2 flashed, turn on your console with D-Pad Down held down on a controller. If you see iplboot output it means PicoBoot itself is installed correctly.

Let me know if you get any output at 2. then we will think what to do next.

Best regards,
Maciej

rakebi, 2022-11-30:
Did put picoboot.uf2, it automatically ejected, got the green led.
Once I try to turn on my console with D-Pad Down held down, nothing happens (green led is on with the console is on).

webhdx, 2022-11-30:
Can you try using analog output? I had one report where digital port adapter would cause PicoBoot not to boot correctly. This is heavily dependent on the adapter and power supply.

rakebi, 2022-11-30:
Yes, switched back to analog but it's not working either.
I also noticed that sometimes it doesn't show the Gamecube animation and instead it directly shows the game/options/memory card/date screen, when it happens it cannot read dvd. When the animation starts, dvd can be played and that works both with analog and digital

webhdx, 2022-11-30:
If you don't get any debug output when holding D-Pad Down it means something happened to the wiring and you need to disassemble the console and check. While at it, post photos of your soldering.

rakebi, 2022-11-30:
I don't think the soldering changed, the console was not dropped and it worked fine until plugging the adapter. I checked the pico board and it look the same as before - tried to pull the wire to see if something on the motherboard was loose and everything its in place.
Since the issue started with the plugging of the adapter - that does drain power from the gamecube, it doesn't have any charging cable - do you think the length of the wires (motherboard to pico board) could affect the pico board? Is a shortage possible because of that?
Would you suggest to cut a little bit on the length of the wires? (the black one is quite short already)

webhdx, 2022-11-30:
If you don't get any debug output it means PicoBoot doesn't work at all, it isn't doing anything. Long wires usually cause no screen at all. I suggest posting pictures anyway, I can't validate your setup without seeing how it was installed.

rakebi, 2022-11-30:
I don't think the soldering changed, the console was not dropped and it worked fine until plugging the adapter. I checked the pico board and it look the same as before - tried to pull the wire to see if something on the motherboard was loose and everything its in place.
Since the issue started with the plugging of the adapter - that does drain power from the gamecube, it doesn't have any charging cable - do you think the length of the wires (motherboard to pico board) could affect the pico board? Is a shortage possible because of that?
Would you suggest to cut a little bit on the length of the wires? (the black one is quite short already)
pics: https://i.postimg.cc/XYTGV808/IMG-20221130-222239-1.jpg
        https://i.postimg.cc/43kyCxz2/IMG-20221130-222747-1.jpg

webhdx, 2022-11-30:
The cables are barely soldered. There are loose strains that can possibly short.

rakebi, 2022-11-30:
Do you think soldering them back would help or something is permanently damaged?
If "reparable", are the wires ok or should I "upgrade" them?
Thanks

webhdx, 2022-12-01:
You just need to resolder them. Judging by dry joints I suppose you're using lead-free solder or not enough heat. I suggest using leaded solder instead, it's much easier to work with.

rakebi, 2022-12-02:
Unfortunately on the second try the gamecube got damaged and broke.
Thank you for your time Maciej