webhdx/PicoBoot #70: Another issue with unofficial board - RP2040-zero

Jrmewam, 2022-12-12:
| Answer some general questions first                             	|  	|
|:-------------------------------------------------------------------	|:------:	|
| Which firmware version are you using?                             	| `v0.2` 	|
| Did you compile it yourself?                             	| no 	|
| Do you use alternative install method not mentioned in the guide? 	| no 	|
| Are you using official Raspberry Pi Pico board?                   	| no 	|
| Did you pay someone to install it for you?                        	| no 	|

Hi there, I'm fully aware you don't offer support for unofficial pico boards, so I won't ask any.

I will buy official board and want to use your affiliate links, but I have following questions : 
- on the [Aliexpress pico link ](https://s.click.aliexpress.com/e/_DkGpnTv), which bundle do you recommand?
- the [SD2SP2 Aliexpress link](https://amzn.to/3PNamBe) is actually an amazon link, Since I guess it's a mistake I'm just letting you know in case you want to fix it.

As I said I won't ask for support, but I'll explain my situation, which might concern someone else.

I bought a [RP2040-pico kit](https://www.aliexpress.com/item/1005004635614406.html) because smaller and USB-C.

I think I did a decent job with soldering  : 

[image]
[image]
[image]

But first, when I ordered the board, I noticed the instructions were wrong and mentionned it to the seller : they putted the 3v3 soldering point to the 5v pin! 
[image]

After that, seller updated instructions : 
[image]

Only after facing no response from pico, I noticed not only they changed the 3v3 soldering instruction, but also all the GP! see screenchots above.

so I changed my soldering points according to pinout diagrams : 
[image]
[image]

Unfortunately, I still don't have any answer from pico. I checked soldering with multimeter, everything seems fine...

Long story short, I'm looking to buy your recommanded pico board :)

webhdx, 2022-12-13:
If the issue is SD card setup you can easily validate that: https://github.com/webhdx/PicoBoot/wiki/Troubleshooting-tips#my-console-boots-to-vanilla-gc-menu

Jrmewam, 2022-12-13:
Just tried and I don't see the output, still booting to vanilla GC menu.
So, it means the issue probably comes from the SD2SP2?

webhdx, 2022-12-13:
No. If you don't get ANY output or a black screen it means the Pico is doing absolutely nothing. You should make sure your board is:
1) functional, find example blink firmware for your board and check if you can get LED blinking
2) check your wiring

Jrmewam, 2022-12-13:
Found a blink firmware on [manufacturer's website](https://www.waveshare.com/wiki/RP2040-Zero#Examples_2), I get zero (no pun) blink. Thanks for the suggestion, I'll go and show that to the seller hoping for a refund. Even tho I don't really believe I'll get any ^^

Jrmewam, 2022-12-26:
Conclusion/ epilogue :
Seller tried his best to avoid refund, but thanks to you and the blink firmware, Aliexpress decided to enforce it.
Now, seller decided to take his RP2040 zero board off the website.

I bought your recommanded pico board, everything's working like a charm!
Thank you once again :)

closed 2022-12-26 by Jrmewam

alimao10, 2023-03-20:
a notable difference the green rp2040 after programming the led lights up the rp2040 zero blue after programming the led does not turn on is there any solution?
but the operation is normal

catb50, 2023-03-20:
webhdx is not supporting other boards then the Raspberry Pi Pico. If operation is normal then that's a success. You can look at the code and make your own varient that triggers the LED for your board 

Here is a hint in the picoboot.c change line "const uint PIN_LED = 25" to "const uint PIN_LED = 16" then recompile that should trigger the internal LED on your board if that's how it is internally setup.

alimao10, 2023-03-21:
> webhdx is not supporting other boards then the Raspberry Pi Pico. If operation is normal then that's a success. You can look at the code and make your own varient that triggers the LED for your [...]

Thanks for the tip but unfortunately I don't know how to program. Could you make a firmware for me? then I would test the led on the chip

VoxelTek, 2023-04-15:
> Here is a hint in the picoboot.c change line "const uint PIN_LED = 25" to "const uint PIN_LED = 16" then recompile that should trigger the internal LED on your board if that's how it is internally setup.

Sadly, it seems the LED on the RP2040-Zero is entirely different in how it's set-up. It's a single NeoPixel LED, and therefore takes a stream of data and displays an RGB colour. So it'd take a bit more to get it working.

@alimao10 
> Thanks for the tip but unfortunately I don't know how to program. Could you make a firmware for me? then I would test the led on the chip

_sigh_ I might get around to making the changes myself, cause I'd like this project to be available on more boards. But as said, the RP2040-Zero board will **not** be receiving guaranteed support from neither myself nor `webhdx`. I can't promise anything, but I figure I might as well try and make a change to the code and recompile. 

Anyway, it's not really necessary, since the RP2040-Zero functions as the modchip perfectly fine otherwise.

fengye, 2023-04-15:
I did some modifications to light up the ws2812 led on rp2040 zero, if interested: https://github.com/fengye/PicoBoot

VoxelTek, 2023-04-16:
Oh, nice work! You've done my job for me lol

alimao10, 2023-04-16:
> I did some modifications to light up the ws2812 led on rp2040 zero, if interested: https://github.com/fengye/PicoBoot

I didn't find the uf2 to download

VoxelTek, 2023-04-16:
@alimao10 that's because it hasn't been compiled. You can follow the steps here: https://github.com/webhdx/PicoBoot/wiki/Compiling-PicoBoot-firmware

alimao10, 2023-04-16:
> @alimao10 that's because it hasn't been compiled. You can follow the steps here: https://github.com/webhdx/PicoBoot/wiki/Compiling-PicoBoot-firmware

I don't have the ability to program, I don't even know where it starts unfortunately :(

fengye, 2023-04-21:
> > I did some modifications to light up the ws2812 led on rp2040 zero, if interested: https://github.com/fengye/PicoBoot
> 
> I didn't find the uf2 to download

Now you have it.

alimao10, 2023-04-21:
I programmed here. led turned on just not tested on gamecube its operation  [https://i.ibb.co/vz1ZBD9/IMG-20230421-201003.jpg](img)

fengye, 2023-04-22:
> I programmed here. led turned on just not tested on gamecube its operation https://i.ibb.co/vz1ZBD9/IMG-20230421-201003.jpg

It is expected behaviour LED is solid red when DMA operation failed

alimao10, 2023-04-22:
> > I programmed here. led turned on just not tested on gamecube its operation https://i.ibb.co/vz1ZBD9/IMG-20230421-201003.jpg
> 
> It is expected behaviour LED is solid red when DMA operation failed

I don't understand DMA ? the rp2040 zero, comes without a program, i put your picoboot and the led turned on. the original program did not light the led

I just didn't install it on gc to test its operation

VoxelTek, 2023-04-23:
> I don't understand DMA ? the rp2040 zero, comes without a program, i put your picoboot and the led turned on. the original program did not light the led

...yes, that's the point. When there's a failure to interface with a GameCube, it shows the red LED. This just proves that the modchip is working as it should, as far as can be identified outside of being installed in an actual GameCube.