webhdx/PicoBoot #73: Any other wiring alternatives??

juankamq, 2023-01-11:
Hey everyone, wanted to ask if there are any other alternatives to install picoboot. I dont have any port in my board as it is heavily cut up. I've tried this one, but didn't work either (not even the debug boot holding down)
[image]
Wondering if there's a way I can install it directly to the ipl?

Thanks!

juankamq, 2023-01-11:
I think i've found an error with that diagram, doing some continuity tests i've found that GP4 from the MX chip gives continuity to a different via than GP4 in the image
[image]
GP6/7 is correct and I don't have a way to test GP5 other than trying.

Will try later and report back!

juankamq, 2023-01-11:
Confirmed, wiring from the first image is wrong. With the second image correction I got it to boot with no issues (I know, it's super messy but i'm testing)
[image]
Will keep this issue open in case someone wants to add to the documentation.

webhdx, 2023-01-16:
I hesitate to support and recommend alternative wiring methods as they are usually much more difficult due to even smaller pads. However there is one good reason for alternative wirings - you don't mess with non replaceable RTC-DOL. Twitter user @grievre_ebooks shared this picture:
[image]

I haven't tested it myself and I can't guarantee it works. I don't support this kind of installation but if you know what you are doing or you don't want to risk breaking RTC-DOL then it might be worth a try. For most beginners I'd still suggest using official installation method.

I thought I'd post it here so people can reference these diagrams in the future.

fenixreign316, 2023-11-30:
Does anyone have a confirmed and tested 5V solder point for either of the above alternatives?  I am trying to adapt the install such that I never need open my GameCube again once my soldering is done, and it is my last stumbling block in the planning phase.

fenixreign316, 2023-11-30:
> > Does anyone have a confirmed and tested 5V solder point for either of the above alternatives? I am trying to adapt the install such that I never need open my GameCube again once my soldering is done, and it is my last stumbling block in the planning phase.
> 
> I have tested both wirings on my comments. But none of them have 5v, just 3.3v that the pi needs. If you want to do this, I'd recommend this one. I have tested it, and has a correction for gp4. The other one is a lot harder cause it goes to the cpu vias. I havent tested the one in webhdx comment ![image](https://private-user-images.githubusercontent.com/96447575/286973740-7ef32e48-858e-4bc1-af24-3206e3759b31.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDEzNzIzODMsIm5iZiI6MTcwMTM3MjA4MywicGF0aCI6Ii85NjQ0NzU3NS8yODY5NzM3NDAtN2VmMzJlNDgtODU4ZS00YmMxLWFmMjQtMzIwNmUzNzU5YjMxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzExMzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMTMwVDE5MjEyM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTg4M2M1ZGZkZjI0ZTlhMjg5NmZiMWRiZmE1ODk5MDE5YTEzYzZiMWNmMjJhYmE1YmMxNjZhNTRlYmI4YTYxNTYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.piBUh-zyfZvBwqmVWCOoKdqrY1OTHwtmCjOA57bCnkI)

I would, but for two things: even @webhdx has acknowledged that powering the Pico using VSYS is ideal.  And secondly, I have seen people having issues with their 3.3V rail being weak and causing many issues all of which were rectified by powering the Pico on 5V either through the USB (not ideal for long term)or on VSYS.  I’d rather not deal with it since I am using a Pico W and the WiFi chipset will draw 3.3V down even more.

juankamq, 2023-11-30:
@fenixreign316 I see what you're asking. You can get 5v from this pins on a gamecube motherboard
[image]
[image]

joeygcpr, 2025-07-01:
> I hesitate to support and recommend alternative wiring methods as they are usually much more difficult due to even smaller pads. However there is one good reason for alternative wirings - you don't [...]

Hello everyone, I'm a couple years late, but leaving this for reference in case someone wants to try Twitter user @grievre_ebooks pads. They worked with no issues, at least for the RP2040 Zero version of Picoboot installed on a DOL 001. Should work for the Pi Pico, but I too would suggest trying the latest official installation method.  Full story for those interested [here](https://www.reddit.com/r/Gamecube/comments/1lnydrt/reviving_a_dol001_and_modding_it_with_wiring/). Thank you!

[image]

[image]

[image]

[image]