webhdx/PicoBoot #74: Allow update while connected

Colrut, 2023-01-16:
Wondering if it is feasible to install a diode in series with the power supply from GameCube to Pico?  Would this prevent reverse current flow when updating the Pico and allow the power to not have to be disconnected to update the Pico?  I am not an electrical engineer so would not know where to start, but perhaps this is worth looking into.

Thank you for all your work on this.

webhdx, 2023-01-17:
Yes this is indeed what you are supposed to do. It's described in detail in the datasheet: ["4.5. Powering Pico"](https://datasheets.raspberrypi.com/pico/pico-datasheet.pdf)

My diagram does not include a diode because that would greatly complicate the installation for complete beginners (trust me). Instead people can just remove the wire during update. There aren't really any software updates and it shouldn't really be a problem. In the future you'll be able to update PicoBoot from the SD card.

Colrut, 2023-01-17:
Thanks for the quick response.  Looking forward to the next stages of this project.

Dalarialus, 2023-02-05:
Would a 1N5817 be an appropriate diode for this?

mibmo, 2024-01-11:
> Would a 1N5817 be an appropriate diode for this?

Seems like it would work to me - did you get to test it? Am very curious if it did indeed work! (I've been messing with adding a switch in series - adding a diode hadn't even occured to me...)