webhdx/PicoBoot #75: Problem Power Supply Internal

dragoncity17, 2023-01-30:
Hi,

First of all, thank you very much for all your work.

I've installed them on several GameCube, and I've noticed something recently that I can't figure out, or it's that I'm running into "hs" internal power supplies.

The one on the right is doing random boots on Swiss... one time it works... one time it doesn't and I get to the GameCube menu.
[image]

And if I maintain "low" and that it does not boot I have no information... ! 

On the other hand if I replace it by the one on the left, no problem ! And this is what I had to do on 2 GameCube already... ! 

So if you have an idea I'm interested ! 

Thanks again ! :)

webhdx, 2023-01-30:
I think the answer is very simple - failing capacitors. I had very similar case in my childhood GameCube. In recapped the power board and it started to boot consistently.

dragoncity17, 2023-01-30:
In fact I forgot to Say that, I changed all capacitif 1000uf 6.3V but no change.... :-/

I have many GameCube and 5 internal power supply have the same problem.... ! o_O

dragoncity17, 2023-01-31:
Or maybe possible to take 5V (not 3V3) for RPI ?

If it's possible tell me where I will test, thanks for your time ! :)

dbzfanatic, 2023-02-02:
RPi can technically take 5v if you connect to VSYS but that is the max supported voltage. There are 5v pins on the controller board and AV out for sure but there's probably one closer, I'm just not sure where.

dragoncity17, 2023-02-03:
Thanks for you answer. if it's possible yes can be an good solution.

dragoncity17, 2023-02-06:
I had again same problem... so I tested your solution "dbzfanatic" and it works ! :)

So 5V to VSYS it's an good alternative ! :)

Thanks again ;)

philectro, 2023-03-26:
seems similar to #32

dbzfanatic, 2023-03-27:
Potentially. It would make sense if there was a brown-out issue that the 5v rail along with the 3v rail, or maybe even just the 5v rail alone, would carry enough energy to supply the pi while it boots even with an intermittent supply.

awkwardbunny, 2023-05-09:
Was having the same issue and was able to solve it by using the 5V rail to either VSYS or VUSB on the controller board (as dbzfanatic and dragoncity17 suggested)
Works with digital out port (I am using retrobit prism hd)
Not sure if it'll work with the "problematic" power supplies as I am using a benchtop supply to power it (my PSU is in the mail)
Note that the LED will probably also turn on while USB is plugged in
[image]

RenaKunisaki, 2023-09-10:
Just chiming in as another who couldn't get it to work without connecting VSYS. I suspect that should be added to the troubleshooting in the wiki, as well as the "hold BOOTSEL on power on to check if the GameCube still works" trick.

Incidentally, I even tried splicing a different 12V power supply with a higher amperage rating in, and it didn't help. So the issue is probably unrelated to the power supply itself, but instead the power regulation inside the console.

(I also had to adjust the laser diode as it wouldn't detect discs, but, I didn't think to check if that worked *before* installing the chip. Probably coincidence, but I wonder if the extra power draw affects the laser too?)

renanbianchi, 2023-09-10:
I had the same problem, managed to get it to work by replacing the caps from the power regulator board + drawing from the 5v on the controller port for the Pico. Works like a charm now. Even the SD2SP2 adapter (which wasn't working along with the Pico only booting sometimes) is working fine now.

BrandonZero93, 2024-12-14:
i need help please someone i have to unplug my bitfunx hdmi adpater everytime and replug it back it to get pico swiss boot to start everytime or load.  how do i get it to load swiss everytime and not vanilla ? do you think its bad caps ?

BrandonZero93, 2024-12-14:
> i need help please someone i have to unplug my bitfunx hdmi adpater everytime and replug it back it to get pico swiss boot to start everytime or load. how do i get it to load swiss everytime and not vanilla ? do you think its bad caps ?

webhdx, 2024-12-14:
> i need help please someone i have to unplug my bitfunx hdmi adpater everytime and replug it back it to get pico swiss boot to start everytime or load. how do i get it to load swiss everytime and not vanilla ? do you think its bad caps ?

This is a hardware problem with Bitfunx adapter. Wire PicoBoot to 5V through a Schottky diode to mitigate the issue. Look up the troubleshooting guide: https://support.webhdx.dev/gc/picoboot/troubleshooting#cause-cheap-chinese-hdmi-adapter