webhdx/PicoBoot #76: Pico W to support Bluetooth in SDK update

catb50, 2023-02-11:
| Answer some general questions first                             	|  	|
|:-------------------------------------------------------------------	|:------:	|
| Are you my GitHub sponsor? | no 	|
| Did you check older issues to see if it was requested before? | yes 	|

With the upcoming release of Bluetooth support in SDK 1.5.0 for the Pico W. Would it actually be possible to unlock the ability to use Bluetooth controllers? Obviously with the help of Swiss. I feel like this is the beginning to something quite amazing in the gamecube scene... 

I was already excited about the wifi functionality for updates and possible FTP server access with Swiss. Having the Pico W board configured via Swiss for internet access could even eliminate the need for the dead internal battery issue by just always syncing time with NTP server upon gamecube boot. Not sure where the hardware limitations start kicking in.

webhdx, 2023-02-12:
I haven't looked into BT stuff yet but it should be possible as we still both cores doing practically nothing. Even if we utilize first core for GC communication there is still second core available. The problem is that it'll require a lot of time to implement and it's hard to justify adding it when we have awesome https://github.com/darthcloud/BlueRetro project. Adding BT controller support to PicoBoot would also require adding more wires to the install.

AceCuba, 2023-03-21:
Hey there man . Frist of all thanks . The blue retro proyect its amaizing but expensive . Im from Cuba and a esp32 its really expensive 40-50usd the chip Alone .if you can implement the bt and wf yo the gamecube using the pico w.  That Will by awesome . Thanks for allí ypur work and bring back a pice of my chillhood

AceCuba, 2023-03-21:
Sorry for my english

webhdx, 2023-03-22:
I understand but this would require spending **months** on my side so I can't really see this happening. I'm sure there are many other people who can try porting BlueRetro to Pico W.

mitchellcairns, 2024-04-12:
https://github.com/HandHeldLegend/PicoBoot-BT

Got it working 🍡