webhdx/PicoBoot #80: emulating memory card with swiss

philivy, 2023-03-06:
| Answer some general questions first                             	|  	|
|:-------------------------------------------------------------------	|:------:	|
| Are you my GitHub sponsor? | no 	|
| Did you check older issues to see if it was requested before? | yes 	|

dear friend, 
I can't emulating memory card with swiss. Is it my  GC harware having issue or normal with 0.2 V picoboot ?? 
thanks

philivy, 2023-03-06:
[image]

webhdx, 2023-03-06:
Hello
First and foremost, you're using wrong board. I only support the official Raspberry Pi Pico boards and do not provide any support
to clone or 3rd party boards.

Second, this is not an issue with PicoBoot because memory card emulation is handled by Swiss. AFAIR it's currently not supported when running audio streaming games from SD card.

Best regards,
Maciej

closed 2023-03-06 by webhdx