webhdx/PicoBoot #82: pico w support

f3bandit, 2023-03-22:
| Answer some general questions first                             	|  	|
|:-------------------------------------------------------------------	|:------:	|
| Are you my GitHub sponsor? | no 	|
| Did you check older issues to see if it was requested before? | yes	|
Can we get a build for the pico w that enables the led, as the led is not
tapped to a gpio pin, apparently it's tied to the wifi chipset some how.
I'm not looking for the wifi to be supported yet as I know that will take
time. Just want to use a pico w to be future proof for when that happens
I don't need to unsolder the pico and solder in a pico w, when I can just
solder once and be done. thx

webhdx, 2023-03-22:
You can use Pico W with the existing firmware. The LED doesn't serve any meaningful purpose at the moment. It won't be future proof because you'll need to change the wiring with new firmware anyway.

catb50, 2023-03-22:
It's just a different definition in the code `WL_GPIO0` vs `GPIO25` I don't think @webhdx will recompile and re-upload just to change a LED light, but will to allow other fun features to take advantage of the Pico W in the future. Open source is available if you wanted to try and change the definition in the code yourself and recompile.

webhdx, 2025-10-28:
Pico W is supported in the latest firmware.