webhdx/PicoBoot #83: I'm missing a port

rdistiso, 2023-03-28:
| Answer some general questions first                             	|  	|
|:-------------------------------------------------------------------	|:------:	|
| Which firmware version are you using?                             	| `v0.1` 	| None
| Did you compile it yourself?                             	| yes/no 	|No
| Do you use alternative install method not mentioned in the guide? 	| yes/no 	|No
| Are you using the official Raspberry Pi Pico board?                   	| yes/no 	| Yes
| Did you pay someone to install it for you?                        	| yes/no 	|No

My GameCube is missing SP2 even though my model is DOL-001

webhdx, 2023-03-28:
You can use SD Gecko adapter:

[image]

It works the same way as SD2SP2, the same performance and such.

rdistiso, 2023-03-28:
Any recommendations for the size and brand? (That isn't SanDisk)

webhdx, 2023-03-28:
I'm using various SD cards and never had any compatibility issues. I'd just get anything that is genuine, avoid buying knockoffs.

rdistiso, 2023-03-28:
One last question, Can I patch my rom of Double Dash and use that because I want to play some mods on real hardware

webhdx, 2023-03-28:
I haven't tested it myself. I'm not sure.

closed 2023-03-28 by webhdx

rdistiso, 2023-03-28:
Thank you for your time