webhdx/PicoBoot #85: Does anyone have the installation diagram for Gamecube DOL-002 BR ?

alimao10, 2023-04-16:
| Answer some general questions first                             	|  	|
|:-------------------------------------------------------------------	|:------:	|
| Which firmware version are you using?                             	| `v0.2` 	|
| Did you compile it yourself?                             	| no 	|
| Do you use alternative install method not mentioned in the guide? 	| no 	|
| Are you using official Raspberry Pi Pico board?                   	| yes 	|
| Did you pay someone to install it for you?                        	| yes 	|

I just wanted a confirmation if the picoboot installation in the BR DOL-002 model is the same as the DOL101 USA

webhdx, 2023-04-16:
I don't own DOL-002 model but I'm pretty sure it has the same motherboard revision as DOL-001. Does your console have SP2 port?

alimao10, 2023-04-16:
> I don't own DOL-002 model but I'm pretty sure it has the same motherboard revision as DOL-001. Does your console have SP2 port?

no have sp2 port

webhdx, 2023-04-16:
I think you should just disassemble the system and compare with the installation instructions. One for DOL-001 or DOL-101 should work.

renanbianchi, 2023-04-16:
It's the same as dol-001.

TheOldOne822, 2023-04-29:
It was my understanding that the BR Camecube was a DOL-001 and the PSU was DOL-002

closed 2023-07-24 by webhdx