webhdx/PicoBoot #87: Trouble installing PicoBoot (DOL-101 USA)

MaltMiltank, 2023-04-26:
| Answer some general questions first                             	  |  	     |
|:-------------------------------------------------------------------	|:------:|
| Which firmware version are you using?                             	| `v0.2` |
| Are you using the official Raspberry Pi Pico board?                 | yes |

I'm trying to install PicoBoot on a 101 I have. Seemingly no matter what I do, I get no output with the Pico connected, though the light on the GC and Pico turn on. I can successfully disconnect everything and have the GC boot like normal, but with the Pico connected, I get nothing. Triple-checked that my connections are correct. I got ahold of a 001 from my sibling and installed it perfectly fine on my first try using the same exact Pico and the same exact firmware, so I'm wondering if I'm missing a step for 101 installation? Pictures below:
[image]
[image]
[image]

MaltMiltank, 2023-04-27:
Update, I got it to work! Turned out I burned the PCB pretty bad under the blue wire connected to the chip. No idea how it booted fine after the first time I de-soldered everything, but after that it appeared the GC was toast. Managed to fix it in the most nasty-looking way possible, (scratching off the solder mask and extending the leg with solder) but it does work now!
[image]

closed 2023-04-27 by MaltMiltank

webhdx, 2023-04-27:
Interesting. Your initial attempt looked really good. I'm not sure why it might have lost connection with the pad. I'm glad you fixed it.