webhdx/PicoBoot #88: Question on extra Pico pins

Bobby-McGonigle, 2023-05-19:
| Answer some general questions first                             	|  	|
|:-------------------------------------------------------------------	|:------:	|
| Are you my GitHub sponsor? | yes/no 	|

No

| Did you check older issues to see if it was requested before? | yes/no 	|

Yes

I'm interested in adding 3 to 6 Neopixel LEDs to my GameCube

- Is there any issue modifying the Pico boot script to add in the Neopixel library, then loop some colors on the device? Or would that impact performance in any way?

- is there enough power to drive 3 to 6 Neopixel LEDs off of one of the Pico pins (between 150ma to 300ma of power needed at max brightness)? Or could this cause an issue to either the GameCube or Pico?

I'm a novice tinkerer and have built with Neopixel a before, but not on anything this complicated

But I think there is plenty of IO, just matter of will I negatively impact the script and if there is enough amperage yo drive the LEDs

Love PicoBoot by the way, I've installed 2 myself, such a great solution!

webhdx, 2023-05-19:
Hey @Bobby-McGonigle 

I don't think there are any reasons why it wouldn't work. I think someone ported PicoBoot to RP2040-Zero board which has Neopixel LED on board. It's just something I'm not interested in doing myself since I'd like to keep PicoBoot as simple as possible. But I encourage people to fork the project and add more features like that.

webhdx, 2023-05-19:
BTW Here is the project I've mentioned: https://github.com/fengye/PicoBoot Its only problem is that it reverts system clock to stock 133MHz which for now doesn't really matter but in the future it will. It can be easily fixed tho - someone just needs to hook up logic analyzer and modify delays to match WS2812 specification.

Bobby-McGonigle, 2023-05-19:
Thank you @webhdx 

I'll give it a try, and I'll post a fork if successful.

Good note about capacitors, I'll see about re-capping before I jump into this.

Thanks!

jbarker2160, 2023-05-19:
I'd run a memory profiler after each time a library is added. Running out
of RAM can completely mess up your timings!

paulmreese, 2023-09-04:
You can return the system clock to its normal setting once all of the PicoBoot code has run, basically where the `tight_loop_contents()` call occurs.
`set_sys_clock_khz(125000, true);`

I've currently got a Pico W running as an Access Point serving a control website. The controls are served as a captive portal, and I've got it fairly hard-coded to run the 5 NeoPixels I installed (4 controller ports and power). I intend on uploading the code in the coming days, but I wanted to say that the power does work for me. I'd also mention that I solder a wire onto the 5V of the controller port board similar to as shown in [this post](https://github.com/webhdx/PicoBoot/issues/75#issuecomment-1539225503) , and that's the source of power for the Pico and the NeoPixels.